subsurface/subsurface #1186: Support for skin temperature with Scubapro G2

p-neves, 2018-04-04:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:
Is it possible to add support for the skin temperature info recorded by the Scubapro G2 computer?
This info is recorded during the dive via the heartband. It will add (yet) another button on the toggle bar, but...

### Operating system:

### Subsurface version:

4.7.7-167

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2018-04-04:
I do worry about diminishing returns vs. overload of the UI / profile.
Not saying "don't do this", just raising a concern...

p-neves, 2018-04-04:
Yes. I know what you mean. The thing is that we can choose what to display by toggling the buttons on the bar... Although it will add another button on the bar...

dje29, 2018-04-04:
+1

I could provide help with implementation and/or test/test data (I'm using G2 with Heart Rate Belt)

satoschu, 2020-05-13:
Hi, the Integration of this would be great! I can also help to test it.
Best regards

mikeller, 2026-10-18:
Hi @p-neves,

This has sustained interest over several years (including a duplicate report, GH#2832, closed in favour of this one) and two users (dje29, satoschu) offered test data/implementation help back in 2018/2020, but that's now a while ago. Are you still interested in this, and if either of those two volunteers are still around, would you be up for helping again? We'd want to coordinate with whoever's actually willing to help get this over the line, since the main gap has been an owner, not technical feasibility.

satoschu, 2026-10-23:
> [Verfolgungsbild][https://github.com/notifications/beacon/AH2SP6CC7I4FGQTK74PBXG35UQ2LRBFCNFSM6AAAAADCPBJGAGWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTYAAAAAC5B6GB5KM4TFMFZW63VHMNXW23LFNZ2A.gif]

Hello,

Yes, I'am still interested in implementing skin Temperatur/Pulse into the P
program.
If it is possible for me I try to help!
Thank you for coming back!

Best regards

Tobias

mikeller, 2026-10-23:
Hi @satoschu,

Thank you. I need to correct my earlier comment: pulse/heart-rate support is already present. Subsurface imports heart-rate samples from the Scubapro G2 path and can show them in the dive profile using the heart-rate control.

The remaining question is skin temperature. I do not see that exposed as a separate value in the current import path, so before treating this as new implementation work, could you please check a recent Subsurface version with a dive recorded using the heart-rate belt?

If heart rate is not imported or displayed, please attach an anonymised `.ssrf` export containing one affected G2 dive and state the Subsurface version and platform. That will let us determine whether this is an import problem or only the separate skin-temperature value that remains unsupported.