subsurface/subsurface #1188: divelogs.de SSL errors

ledvinap, 2018-04-05:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Attempts to download logs  from divelogs.de (probably any communication) fails with `Error: SSL handshake failed`

divelogs.de provides certificate for `*.divelogs.de`, this certificate is invalid for `divelogs.de` hardcoded in subsurface.

https://divelogs.de/xml_available_dives.php has certificate error in Chrome, https://en.divelogs.de/xml_available_dives.php has valid certificate and returns expected error (login failed)

Simply replacing  `divelogs.de` with `en.divelogs.de`  may fix the problem.

### Operating system:

Tested on Windows 7 x64

### Subsurface version:

4.4.7

### Steps to reproduce:

Import-Import from divelogs.de - Download. Always fails, username/password has no effect

### Current behavior:

`Error: SSL handshake failed`

### Expected behavior:

Logs imported

### Additional information:

### Mentions:

dirkhh, 2018-04-05:
I assume you mean Subsurface 4.7.7 ?
I've reached out to the divelogs.de maintainer as this sounds like it's a misconfigured certificate on his site...

dirkhh, 2018-04-05:
Rainer already responded. He replaced the certificate. Please test again, but I think this should be fixed.

closed 2018-04-05 by dirkhh

ledvinap, 2018-04-05:
Works now, thanks!