subsurface/subsurface #1200: [Bug][Planner] GF are not saved

fcgrims, 2018-04-10:
[Bug] GF (Gradient Factors) in Buhlmann deco in the planner are not saved.

### Issue long description:
Planning a dive.
Modidy GF, for example 80/80.
Profile is update accordingly.
Save.
Select another dive and  select again the previously planned dive.
Profile is shown with 45/75 GF <-- here is the bug !
Re-open it with planner (GF are set to chosen GF, in the example 80/80)

### Operating system:
Last Mac OS High Sierra

### Subsurface version:
Subsurface 4.7.8 official release.
It happens with previous version.

neolit123, 2018-07-08:
@atdotde do you think this is an easy fix and can be included in 4.8.1?

atdotde, 2018-07-09:
This is not exactly a bug. Subsurface has two settings for the GFs, one for showing dives in the logs (and then also in the profile plot) and one for the planner. To see your planned dive in the divelist with a 80/80 ceiling, you need to set the gradient factors in the preferences.

closed 2018-07-09 by atdotde

fcgrims, 2018-07-09:
Planner : so you cannot plan different dives with different GF. Nonsense. The preference is for setting GFs by default, but one can adapt GFs for each planned dive, and one can retrieve these values at the next opening in the graph, and not the values by default (preferences!).

Log : same kind of bug here. a logged dive comes with its own GFs values. They are recorded. But the name of the main window is made of the defaults GFs values (preferences). Therefore, the user do not know what GFs are used for drawing graph of tissues. Therefore the user have a lack of confidence in the whole software...

dirkhh, 2018-07-09:
I respectfully disagree with your "analysis". The GF shown above the profile is the GF that Subsurface uses to calculate the ceiling that is shown in green.
Subsurface typically doesn't know which algorithm the dive computer used (for a couple of them we have access to this information, for most of them we don't).
This works as designed and I see no reason to change that.