subsurface/subsurface #1204: Subsurface Mobile UI: Cloud sync / Offline mode wrong sidebar text

Bocio, 2018-04-16:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Opening the sidebar menu "Dive management"  when the device is connected with the cloud AKA automatic cloud synchronization is enabled I get:

[image]

The last item correctly indicates me that if I press "Offline mode" I will disconnect from the cloud. In other words, text string shows the new status I will go if I press that button.

Pressing the "offline mode button" I get no message and the UI become:

[image]

As you can see there is an error because the Auto sync is not enabled now but if I will press the button I will enable the Auto cloud sync.
The correct text would be "Enable auto cloud sync". So it will be consistent with all menu items: the button text will always shows the action, not the current state.

There is another small error because now (I'm offline) when I press the "Auto cloud sync enabled" button the UI will show me the "offline mode" button but I get a completely wrong popup message:

[image]

It alert me that I'm going to disable auto sync which is not. This popup should appear when I press "offline mode".

Of course we could choose to indicate the current status on that buttons but it will break the overall meaning of other menu items. I guess the error arose because the short "offline mode" text string is misleading by itself. Maybe a more verbose like "Disable auto cloud sync mode" would be better suited along "Enable  auto cloud sync mode". The popup should be changed anyway.

### Operating system:

Android all version

LG X power, Nexus 5x, Nexus 7

### Subsurface version:

Subsurface Mobile 2.0.3 (4.7.8.49)

Official and test

### Steps to reproduce:

Already in the long description
### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2018-04-16:
Thanks for the detailed report including screen shots. So much easier to understand what's going on!
I'll look into it as I am currently working on the mobile app, anyway :-)

Bocio, 2018-04-16:
It's easy to spot if you play a bit with the UI but if by chance your device is offline, on the sidebar you see "Auto cloud sync enabled".  Digging on the log I found a suspicious string "have cloud credentials, but user asked not to connect to network". So I discovered the bug.

Bye

dirkhh, 2018-04-16:
I changed the two texts to be symmetric:
Disable auto cloud sync
Enable auto cloud sync
now I need to find that popup :-)

commit f8742a48 (2018-04-16): QML UI: show offline notice when going offline

The logic was backwards and showed the notice when the user switched to
auto sync mode.

Fixes #1204
closed 2018-04-16 by dirkhh