subsurface/subsurface #1208: iOS, date formatting/editing not working

janiversen, 2018-04-21:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Edit the date/time field on macOS works, but edits done on iOS are discarded. 

When I add a dive on iOS, it does not obey the format selected on my iPad, not the one set in preferences in subsurface, macOS. I use dd/mm/yyyy as standard, but on iOS it adds mm/dd/yyyy (American standard).

### Operating system:

macOS 10.13.4 () and iOS 11.3

Mac with intel processor
iPad Pro and iPhone 6

no, macOS works

### Subsurface version:

4.7.8 (macOS, downloaded from subsurface site)
2.0.1 (iOS, downloaded from appleStore)

do not know.

official release

### Steps to reproduce:

1) Add a dive in iOS
2) look at the date/time format
3) modify e.g. the date, but keep the format
4) save
5) view, date not changed
6) add date in standard dd/mm/yyyy format
7) save
8) view, date not changed
9) sync, and view on macOS
10) modify date on macOS
11) sync, and view on iOS (WORKS).

### Current behavior:

Date format mm/dd/yyyy, non-editable

### Expected behavior:

Date format dd/mm/yyyy (or what is set in iOS), editable

### Additional information:

### Mentions:

mfbernardes, 2018-05-14:
@janiversen Can you try again with the latest version from FlightTest? Preferably with a clean install.

janiversen, 2018-05-14:
Works in flightTest (but there is a formatting problem, will file a seperate issue, since it is more general)

closed 2018-05-14 by janiversen