subsurface/subsurface #1211: Editing duration and save, dive listbox is not updated

janiversen, 2018-04-21:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

on Mac, when updating the duration and saving, the listbox below is not updated.

### Operating system:

macOS

Mac

### Subsurface version:

4.7.8

Official release, download from HP

### Steps to reproduce:

1) view a dive
2) update duration of dive
3) save
4) check listbox of dives, column Duration is not updated

### Current behavior:

See above

### Expected behavior:

Duration column in listbox for the edited dive should be updated.

### Additional information:

### Mentions:

atdotde, 2018-04-21:
Sorry, I don't get it. How exactly do you change the duration of a dive? Neither the field in the dive list nor the one in the statistics tab should be editable. In fact, the duration is computed from the profile. Or do you change the profile (for example via "edit dive in planner"?

janiversen, 2018-04-22:
When you add a dive manually you need this field (it is editable in the iOS version as well, but in minutes, and that works) because date/time signals start of the dive, and there are no date/time to end the dive. For sure if you upload a log/profile it will be calculated. See the attached screenshot.
[image]

dirkhh, 2018-04-22:
This used to work - I'm trying it here locally with current master and it seems to work just fine. Unless you start your number with a '5'. Try entering 40 and you'll see it work. Then try 50 and it fails. WTF? OK, something to look into.

janiversen, 2018-04-22:
Confirmed, I would never have guessed that 5 is a "dirty" number :-)

I am happy to help, but would beed a code pointer

dirkhh, 2018-04-22:
desktop-widgets/tab-widgets/maintab.cpp on_duration_textChanged() calls parseDurationToSeconds(text) and my guess is that that's what's broken.
This code was written by me and therefore is likely to be completely bogus.
Staring at it for 30 seconds I can't quite tell why this doesn't work, but my suggestion would be to write a few unit tests to make sure that it parses strings as intended and see what breaks.
(or my quick analysis could be wrong and the error could be higher up in the stack - but this is where I would start)

janiversen, 2018-04-22:
Let me know if I can help with information. I hope to be close to have the Mac version up and running, due to other Open Source projects, I have to do it without homebrew, and I try to document every step, so I am a bit slower than normal.

dirkhh, 2018-04-22:
if you tell the build script to -build-deps then you should be able to do this all without Homebrew.
I need to do that in order to build binaries that others can use (as this allows me to build against an older macOS - Homebrew by design always builds for your local machine (what an incredibly self centered approach))

janiversen, 2018-04-22:
thanks, one more thing to document, but something that makes life easier for Mac users.

bstoeger, 2018-05-04:
The problem is how dive duration is transported via samples which are transformed back by `fixup_dc_duration`.

For 40 min I get:
```
                SAMPLE 0: time: 0 depth: 0
                SAMPLE 1: time: 180 depth: 15000
                SAMPLE 2: time: 2040 depth: 15000
                SAMPLE 3: time: 2160 depth: 5000
                SAMPLE 4: time: 2340 depth: 5000
                SAMPLE 5: time: 2406 depth: 0
```

For 50 min I get:
```
                SAMPLE 0: time: 0 depth: 0
                SAMPLE 1: time: 180 depth: 15000
                SAMPLE 2: time: 380 depth: 0
```
Which makes no sense whatsoever.

In any case even if this is fixes, the whole duration->samples->duration business seems questionable for manually entered dives. You will get different numbers out than what you put in. I could imagine this being the root-cause of strange bugs such as #554.

My personal opinion would therefore be: don't touch ("fix") duration, depth, etc. in manually entered dives. Just leave them as the user entered them.

bstoeger, 2018-05-04:
Wow. There is some major confusion going on.

The reason that the profile is truncated is the following loop-condition in `qt-models/diveplannermodel.cpp` (function `loadFromDive`):
```
              if (dc->last_manual_time.seconds && dc->last_manual_time.seconds > 120 && lasttime.seconds >= dc->last_manual_time.seconds)
                      break;
```
The funky thing: The variable `dc->last_manual_time` is set *after* `loadFromDive()` in `create_dive_from_plan()`.

So what happens:
1) You clear the field. `dc->last_manual_time` becomes 0. The if will not trigger, everything ok.
2) You enter the first digit, everything OK, because still 0.
2a) You entered a digit <= 4 -> `dc->last_manual_time` becomes <= 120, the if does not hit, no probs.
2b) You entered a digit >= 5 -> `dc->last_manual_time` becomes >= 150, the if hits, disaster ensues.

Removing the above if condition fixes the problem, but it is probably there for a reason. I think there is no sense in me continuing to debug this. @atdotde @mturkia, I think this is your call.

BTW: With the if removed, I managed to DOS the application by entering 555555 in the duration field:
```
Polygon too complex for filling.
Polygon too complex for filling.
Polygon too complex for filling.
Polygon too complex for filling.
Polygon too complex for filling.
Polygon too complex for filling.
Polygon too complex for filling.
Polygon too complex for filling.
Polygon too complex for filling.
Polygon too complex for filling.
```

bstoeger, 2018-05-04:
@willemferguson: The `last_manual_time` stuff seems to be your code. Please check.

commit fc7cb09b (2018-05-07): Planner: Correctly fill out last_manual_time in fake_dc()

By not filling out this value, entering of manual dives was broken
for dive lengths starting with a digit 5 or higher.

Fixes #1211
closed 2018-05-07 by neolit123