subsurface/subsurface #1212: CSV Import, missing fields "Ratings", "Visibility", "SAC"

janiversen, 2018-04-22:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When trying to import a CSV file, most of fields can be selected, but at least
"Ratings", "Visibility" and "SAC" cannot be assigned (see screen shot). 

### Operating system:

macOSX 10.13.4

Mac

### Subsurface version:

4.7.8

Official release

### Steps to reproduce:

1) select Import/Import log files
2) choose attached file
3) Try to find assign a "blue" field to the visibility, quality, SAC columns
[image]

### Current behavior:

Not posible to import the 3 values

### Expected behavior:

Possibility to import the 3 values.

### Additional information:

See dives.csv (added .txt to get it uploaded)

[Dives.csv.txt](https://github.com/Subsurface-divelog/subsurface/files/1935286/Dives.csv.txt)

### Mentions:

dirkhh, 2018-04-22:
SAC is a derived value in Subsurface - so depth/duration/gas consumption are used to calculate SAC rate.
Visibility and rating my not have been added as input fields by @mturkia , yet

janiversen, 2018-04-22:
If it is properly calculated, no need to add it....but Visibility and Quality are really missing.

Again I can help, if I have a code pointer (that is my way getting into the sources)

dirkhh, 2018-04-22:
That's why I tagged Miika above - all of the parsing code is his baby and that code is, errr, interesting :-)
xslt is not for the faint of heart.

commit 12e4b6d6 (2018-05-01): CSV import: include visibility and rating in import

Add support for visibility and rating to CSV import dialog.

Fixes #1212
closed 2018-05-01 by neolit123