subsurface/subsurface #1213: Not possible to edit dive information from dive list.

janiversen, 2018-04-22:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When viewing "dive all" it is possible to edit most of the fields for an uploaded dive, which is nice.
When viewing "dive list" and trying to edit a dive an error dialog box occurs, this is inconsistent behaviour.

### Operating system:

macOSX 10.13.4

Mac

### Subsurface version:

4.7.8

Official release

### Steps to reproduce:

1) select "view"/"all"
2) Edit e.g. buddy, it works
[image]

3) select "view"/"dive list",
4) choose an uploaded dive (NOT manually entered)
5) select "log"/"edit dive"
6) you get an error dialog box
[image]

### Current behavior:

Error dialog box

### Expected behavior:

Change to "all" screen"

### Additional information:

### Mentions:

dirkhh, 2018-04-22:
I see your concern, but I will file this under "Works as Designed". The dive list is designed to be read only.

janiversen, 2018-04-22:
No problem, but then "edit dive" should not be selectable in the menu :-)

dirkhh, 2018-04-22:
True - we should gray that out in this mode. Reopening the issue...

bstoeger, 2018-05-03:
I tried to enable "Edit dive" for all dives: https://github.com/bstoeger/subsurface/tree/edit and removed a few lines of code-duplication while at it.
The thing is ... this puts the application in edit mode, but apart from that is pretty much a no-op.

Therefore another idea: Say "Edit dive profile" instead of "Edit dive" and enable only if a single, planned or manually entered dive is selected?

janiversen, 2018-05-03:
> I tried to enable "Edit dive" for all dives: https://github.com/bstoeger/subsurface/tree/edit and removed a few lines of code-duplication while at it.
> The thing is ... this puts the application in [...]
Seems I missed something because you also need (at least some fields) to edit a downloaded dive, so why not keep “Edit dive” and allow edit all (planned etc) or limited ?

rgds
jan i

bstoeger, 2018-05-03:
The thing is that for downloaded dives "Edit dive" is a no-op. You can start filling out the fields at any time anyway.

janiversen, 2018-05-03:
> 
> The thing is that for downloaded dives "Edit dive" is a no-op. You can start filling out the fields at any time anyway.
I agree editing the downloaded data is a no-go, but “edit-dive” could open for all editing (planned... dives), and open the extra fields if it is a downloaded dive. Both operations are “edit dive” seen from a user perspective.

rgds
jan i

bstoeger, 2018-05-03:
Perhaps #1236 works for you. But this will need some discussion.

atdotde, 2018-05-03:
Hi,

> On 3. May 2018, at 17:12, bstoeger <notifications@github.com> wrote:
> 
> The thing is that for downloaded dives "Edit dive" is a no-op. You can start filling out the fields at any time anyway.
> 
> 

„Edit dive in planner“ does work for downloaded dives and might be more what you are looking for.

Robert

bstoeger, 2018-05-03:
If I understand the bug report correctly, editing the profile is *not* the point.

The perceived problem is that "Edit dive" gives an error message instead of doing what it says: editing the dive. I don't really have an opinion what to do in this case.

bstoeger, 2018-05-07:
@janiversen: Can you please check if #1236 works for you?

janiversen, 2018-05-07:
Done, works, see detail in PR. Nice job.

janiversen, 2018-05-08:
solved with #1245, thanks.

commit 57bf174c (2018-05-13): Desktop: On dive edit from the dive list or map, switch to new state

If "Edit dive" is selected from the dive list or the map view, switch
to a new mode, which shows the dive infos and the profile.

After the edit, switch back to the previous state.

Fixes #1213
closed 2018-05-08 by janiversen