subsurface/subsurface #1214: qmake is no longer supported, but still needed in build.sh

janiversen, 2018-04-22:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

INSTALL says:
"As of Subsurface 4.5 we have switched our build system to cmake.
qmake based builds are no longer supported."

but build.sh still needs qmake to be available (line 139...):
# find qmake
if [ ! -z $CMAKE_PREFIX_PATH ] ; then
        QMAKE=$CMAKE_PREFIX_PATH/../../bin/qmake
else
        hash qmake > /dev/null && QMAKE=qmake
        [ -z $QMAKE ] && hash qmake-qt5 > /dev/null && QMAKE=qmake-qt5
        [ -z $QMAKE ] && echo "cannot find qmake" && exit 1
fi

So we either need to update the text in INSTALL or remove the need to use qmake
(last part should be preferred).

### Operating system:

MacOSX 10.13.4

Mac

Yes that part of build.sh is universal

### Subsurface version:

4.7.8

Source tree

commit 4186733740b1b73a762a9995bf36099051bf6dd5

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2018-04-22:
Sort of kinda yes and no
So Subsurface (the desktop app) now uses cmake.
But for stupid reasons that make me cringe, Subsurface-mobile for iOS uses qmake.
And regardless, the googlemaps plugin requires qmake as well.
So I think that text should just be removed as it is confusing. Or it should be replaced with something like "on all platforms except for iOS Subsurface requires cmake"

janiversen, 2018-04-22:
I am sorry to say it, but I try to build on macOSX, and build.sh kept complaining that I did not have qmake installed.

I presume you are right, and it is only needed for iOS (something that WILL change soon), but then we miss an "if iOS" in build.sh

dirkhh, 2018-04-22:
No, read what I said.
The googlemaps plugin requires qmake

janiversen, 2018-04-22:
UPS !! my error, I will adjust my proposal for INSTALL (it is not filed yet, it waits until I have successfully build for Mac and iOS).

dirkhh, 2018-04-22:
see my comment in the other issue about using
./subsurface/scripts/build.sh -build-deps
on Mac.
There is another wrinkle on Mac and Windows - the official Qt binaries no longer include WebKit, so unless you provide your own QtWebKit, your personal build will be lacking Facebook integration and Printing support...

janiversen, 2018-04-22:
hmm I just installed 5.10 and it included webkit. Just wondering, do we have the baseline (e.g. minimum Qt 5.4) documented somewhere ?

janiversen, 2018-04-22:
hmmm just controlled it is called "Web engine" will see later if it is the same as the good old "Qt webkit"

dirkhh, 2018-04-22:
No it is not. WebKit is the old KDE browser engine that was adopted by Apple and is the basis of Safari. WebEngine is based on the Chrome backend and COMPLETELY different. We have the ability to build part of our features against WebEngine, but not the printing code.
You can indeed very easily build QtWebKit against whatever Qt version you install.
As you would say "something we should document" :-)
Also, to your earlier question, I believe we require at least Qt 5.5 (we use some APIs that aren't in 5.4) and we recommend 5.8 or newer. Especially if you want BLE support 5.9.2 or newer is highly recommended.

neolit123, 2018-04-22:
> We have the ability to build part of our features against WebEngine, but not the printing code.

WebEngine cannot be built with MINGW.