subsurface/subsurface #1216: source code should not use depreciated QtWebKit

janiversen, 2018-04-22:
### Describe the issue:

- [ ] Bug
- [x ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

We recommend using Qt 5.8, QtWebKit was depreciated in Qt5.6, therefore it seems logical to update the code to not use QtWebKit but the never components.

### Operating system:

any

### Subsurface version:

4.7.8

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

atdotde, 2018-04-22:
Unfortunately this is a know issue. We have transformed most code to use WebEngine but printing still depends on it. The problem is the page breaking: We render the pages in html but since html has no idea about breaking on different pages, we have to modify page elements to fit on individual pages. In WebEngine, however, you don't have access to page elements, you are supposed to do that in JavaScript. It would be great if you started to work on this but for the time being I will close this ticket.

closed 2018-04-22 by atdotde

neolit123, 2018-04-22:
the main problems is that WebEngine cannot be built with MINGW.

dirkhh, 2018-04-22:
BTW - Qt is now in a semi-schizophrenic mind on WebKit and WebEngine. They officially tell you that (a) included in Qt is only WebEngine and (b) that QtWebKit still works and is supported by a different developer team. So if you simply download QtWebKit 5.212 this works perfectly fine. Clone the sources, configure, install, done.
So at this point I am perfectly happy to do just that... the advantage is that I can build that on Linux, Mac, Windows - and Printing works as designed, etc. Seems like a straight forward trade-off to me.

And again, <1% of our users ever look into compiling from source.
What we need to do is improve the instructions so someone who wants to do that can succeed.
That's trivial on Linux (as the distros do the same thing I do, just build WebKit). NO ONE builds for Windows (there are something like 4 people total who appear to have done this, all of them core Subsurface developers). Mac is the borderline area... there seem to be enough people who want this, so I'm happy to help getting the INSTALL instructions right.

janiversen, 2018-04-22:
Ok, I have a lot of modification to INSTALL, so I will simply add QtWebkit to that.