subsurface/subsurface #1218: Delete button disfunctional on iOS

atdotde, 2018-04-23:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

In the current (testflight) version on iOS, the delete button seems to be dysfunctional. Most of the time, nothing happens when I press it (as if I were missing the button), sometimes the UI reacts (including showing the undo button) be the dive does not disappear from the dive list.

### Operating system:

iOS current

iPhone 8plus

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mfbernardes, 2018-05-14:
Does it happen to work the first time and then gets dysfunctional? or is like you described from the start?

Does the + (add dive) button work correctly?

Do you have lots (a few hundred+) of dives on the dive list?

janiversen, 2018-05-30:
just tested with testflight and my test user (50 dives). It works perfect for me except I have yet to see the undo botton.

janiversen, 2018-05-30:
ups I tested on a iPad pro

dirkhh, 2018-06-18:
What's the state of delete button and undo button on iOS these days. I didn't bring any iOS devices and testing things like this on the simulator typically isn't as representative...

janiversen, 2018-06-18:
It works for me, just tested with my test user, but behaivour is not what I expected (might still be correct).
- press delete button
- divelist is presented, with undo button for less than 1 sec (very short for a user to react)
- divelist is correctly updated 

will check the undo button timeout

janiversen, 2018-06-18:
false alarm the undo timer is ok. So this can, in my opinion, be closed

mfbernardes, 2018-06-18:
The state is works for me :)

The existing Undo button seems to work fine.

There is no Undo button when deleting directly from the dive list (click and hold, then click trash icon). Inconsistent but no big deal and what is there works.

dirkhh, 2018-06-18:
so let's close

closed 2018-06-18 by dirkhh