subsurface/subsurface #1232: Cylinder, does not show in iOS.

janiversen, 2018-05-03:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

As seen in the 2 screenshots, on macOS Cylinder is shown, but not on iOS.

### Operating system:

macOS and iOS

### Subsurface version:

4.7.8

yes master

official

### Steps to reproduce:

have a drive on Mac, where cylinder is present,
view on iOS

### Current behavior:

### Expected behavior:

Cylinder on iOS to be filled

### Additional information:

[image]
[image]

Renamed as .ssrf.txt to be able to upload it.
[Dives.ssrf.txt](https://github.com/Subsurface-divelog/subsurface/files/1971360/Dives.ssrf.txt)

### Mentions:

mturkia, 2018-05-03:
iPhone (and iPad) show only the cylinder type. That information is missing from your cylinder configuration.

janiversen, 2018-05-03:
so in other words it is one more difference in the ui.

we should add the missing field to the ui.

mturkia, 2018-05-03:
I disagree on this as I am using iPhone with quite a small display. Having the screen filled with more data than it can sensibly show will ruin the usability. The mobile version is not meant to be full blown Subsurface but something that you can use to get enough information when not having a laptop available.

janiversen, 2018-05-03:
I highly agree, that if we keep the current layout, all information cannot be displayed. I normally use an iPad, and on that I would really like to see all data, so I avoid starting my Mac.

I am used to having different layouts and different levels of fields included, depending on the screen size, and that is something relatively easy to implement. For smaller screens we could do stacked screens (pretty much like "all" with its tab on a Mac).

This might a good comprise, keeping your iPhone clean and my iPad with all info.

janiversen, 2018-05-03:
and btw. the reason I do no set the cylinder type, is because here in Spain, nearly all recreational dive centers use ALU.

dirkhh, 2018-05-04:
I am confused why that would prevent you from adding AL80 (or AL72, or whatever) as cylinder type.
But to go back to the other question about screen sizes and different layouts... so far we have struggled with finding enough developers with enough time and an inclination to work on QML code to really invest more into the mobile UI. If you want to work on that that would be great - but since the code is shared with Android we'd need to make sure that this is well tested over a wider range of devices.

dirkhh, 2018-05-04:
all that said, at this point I'd call this "works as designed".

closed 2018-05-04 by dirkhh