subsurface/subsurface #1240: Can't select Cressi (Leonardo) on mobile beta - Android

StephenGoodall, 2018-05-06:
Should there be an option for the Cressi Leonardo on the download screen?

I'm using a DiveMate fusion IR, was hoping to see if the FTDI would work but I can't select my model from the list.

I had a look on my windows laptop and it seems to use the FTDI in device manager, I wanted to use PortMon, following the instructions from libdivecomputer for the contribution section on transfer protocol but it didn't run in windows 10.

If there's anything I can try to capture on iOS or windows that might help get the fusion IR working in subsurface let me know, comfortable with command lines :)

StephenGoodall, 2018-05-08:
Just as an update, the intent part is working as it popped up asking if I wanted to open subsurface for this device when I plugged in.
I just don't see Cressi here:
https://photos.app.goo.gl/JkhSvagtu96nSGYNA

StephenGoodall, 2018-05-09:
I think it might need adding to this file:
subsurface/core/downloadfromdcthread.cpp

around line 75, maybe add this (not sure if all these would be supported):

mobileProductList["Cressi"] =
		QStringList({{"Drake"}, {"Edy"}, {"Giotto"}, {"Leonardo"}, {"Newton"}});

StephenGoodall, 2018-05-09:
I've never developed anything for mobile (I've only really used Java for a Tomcat server), but if there are any links to any instructions for how to build it and test it then I'd be more than happy to give it a try. My phone is not rooted though so not sure if I could run it on there to test it out properly

dirkhh, 2018-05-09:
I would be very surprised if that works.
Either way, we will certainly not add this to the official Subsurface-mobile version as this would be rather misleading as Android doesn't support IRDA devices.

dirkhh, 2018-05-09:
I doubt that it works, but I'm not sure. But I'm not going to add a feature to Subsurface-mobile that only works if users buy the DiveMate product - that would seem odd to me.

StephenGoodall, 2018-05-09:
That's fair enough, it shouldn't rely on a specific piece of hardware that is essentially from different project, I was thinking it might also make it work with the Cressi interface, as that would present with the same VID and PID - I don't have one to try out though.
I'll get the relevant tools installed and see if I can build an apk on my laptop to try it out, feel free to close this if it's not something that would be included anyway.

p.s. thanks for all the work on subsurface so far, it's definitely my preferred software! I can always use my laptop to import from the DC

StephenGoodall, 2018-05-11:
So i managed to get it to build the apk, with a bit of help from @neolit123 
It said this in the logs:
[image]

But I'm not sure if I built it from the right branch......is there a mobile beta specific branch I should be building from? I just used master and made the above change I mentioned (to show cressi in the list). It seemed to hit this method in the serial ftdi class though and that kind of stopped it in its tracks :)
```
static dc_status_t serial_ftdi_get_transmitted (ftdi_serial_t *device)
{
	if (device == NULL)
		return DC_STATUS_INVALIDARGS;

	// This is not possible using libftdi. Look further into it.
	return DC_STATUS_UNSUPPORTED;
}
```

Is it worth me trying a different branch/commit? In the manual it mentions:

> On Android devices these are typically dive computers using an FTDI interface using a USB OTG cable (but this is blocked on some Android devices by OS settings).

Would the above log indicate that it is being blocked? Or could I just be missing something?

dirkhh, 2018-05-11:
The screen shot tells me that you are on an Oppo phone. Those have been the one brand that is broken in a lot of subtle ways with Subsurface.

We had to re-architect the way we talk to libdivecomputer about a month ago because many of the APIs changed - it's possible that FTDI on Android got broken and no one noticed (because very, very few people use this and I, for one, can't test it because all of my Android devices refuse to let me access the device nodes).
@glance- have you had a chance to test that this still works?

StephenGoodall, 2018-05-11:
I quite like the font.....guess that's what gave it away! It's a OnePlus 3, it's not rooted.

I don't remember getting a pop up to allow access to the USB (unless it was a long time ago), just the location and storage from when it was installed. Not sure if that helps or not.

dirkhh, 2018-05-11:
I won't comment on the font.
If you didn't get a question whether the FTDI device should be connected to Subsurface-mobile AS YOU PLUGGED IT IN (not when installing the app), then this has no chance of working at all.

StephenGoodall, 2018-05-11:
That one did pop up, when I select it, subsurface opens as it should, so I know that part is working in the sense that it is reading the VID and PID correctly as the FTDI device with the Cressi Leonardo values

glance-, 2018-05-13:
I've tested the ftdi code on Linux, And it works.

Why it didn't work on your device can be due to how the selinux rules got changed a couple of android releases ago. Since then we can't use straight libusb to find devices any longer, because selinux gets in the way of using the normal kernel api's.

The real solution is to patch libusb to call out to the android apis, the same way we're doing to "open" the device.

The other potential solution is to replace the libftdi based driver with the "android" ftdi driver, which talks the android apis directly. The downside is that this way won't help the other libusb based drivers, like the eon steel.

StephenGoodall, 2018-05-14:
Thanks for having a look @glance- 
It's beyond my knowledge but it looks like a few people are working on patching LinuxUSB:

https://github.com/PointCloudLibrary/mobile/blob/master/3rdparty/android/patches/libusb.patch

https://github.com/libusb/libusb/pull/242

It looks like @dirkhh had already commented non the second one last May but it hasn't been merged in yet.

If I build libusb from that branch and replace the libusb-1.0.20 folder, will i be able to build the APK again and test that version out myself?

StephenGoodall, 2018-05-14:
I think, even with the version from that PR, it would require some changes to the subsurface code, is there any chance you could point me in the right direction for which file to look at for that? Or is it going to be spread out across a few classes?
The comment detailing these changes are:

1. You have to use the Java Android API to open a file descriptor to the USB device before you can use libusb.

2. As a regular user, you cannot use libusb's enumeration functions to get a libusb_device because you cannot read sysfs and /dev/bus/usb.

3. After obtaining a libusb_device_handle, you can call libusb_get_device() to get a libusb_device which can be used to get all descriptors you want. No need for additional trickery with the Android API.

4. You must of course properly clean up the libusb_device_handle by calling libusb_close() when you are done.

glance-, 2018-05-14:
Those libusb patches are just other variants on the one i wrote:
https://github.com/Subsurface-divelog/subsurface/blob/master/packaging/android/patches/libusb-android.patch

The inspiration for the patch can be found in the git history.

Not to be rude or anything, but if you need more pointers to know how to hack libusb into working with the new selinux rules, you're not the person to do it.

StephenGoodall, 2018-05-14:
It's not rude if it's true!
I was hoping to be able to add something rather than just be one of those users who say something doesn't work but doesn't try to contribute....but this is well beyond me!
I've only really got minimal java experience :(

Let me know if there is anything I can help with.

dirkhh, 2018-05-14:
It's still rude, though, even if true.
The issue is - we REALLY need someone to do this, @glance- doesn't have the time and no one else here (including myself) is comfortable with this code.
I have not a single FTDI based dive computer, so my motivation to learn how to do this is rather small (even ignoring the appalling lack of time). So I had been hoping that someone with a personal motivation to work on this would step up and figure it out. So far this hope has not worked out.

StephenGoodall, 2018-05-18:
I was playing around with a few apps from the PlayStore and this one seemed to be able to communicate with the FTDI device, it sent a message to the dive computer and received something (not that i could understand it - i had no idea what to send!)
https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal&hl=en_AU

The source code has an FTDI driver (which I think I was using) but it's all java, so I'm not sure if it is of any use or not on this c++ project:
https://github.com/kai-morich/usb-serial-for-android/blob/master/usbSerialForAndroid/src/main/java/com/hoho/android/usbserial/driver/FtdiSerialDriver.java

It seems to be LGPL licensed.

In the meantime, I'm looking into options to root my phone and then I can try this with SELinux set to permissive - but that might not happen as I need to get a full backup first somehow, for if i want to un-root and go back later.

StephenGoodall, 2018-05-21:
Sonic rooted my phone and tried the app in permissive mode, still no joy. Probably from that unsupported message in the ftdi class.

If you need me to test anything in permissive mode I'm happy to give it a try, if it helps with figuring out what's going on

glance-, 2018-05-21:
Can you post the log files so I can take a look?

StephenGoodall, 2018-05-22:
Bear in mind that this is with my modification to the downloadfromdcthread.cpp file, to add cressi to the list. I also set FTDISUPPORT to ON in the cmakelists file - I have no idea what I'm doing with that so it might not make any sense (or difference) to do that! Happy to change it back and rebuild the apk if you prefer it to be clean?

option(FTDISUPPORT "enable support for libftdi based serial" ON)

https://photos.app.goo.gl/0ICDAuuneK9NaleL2

https://photos.app.goo.gl/7ocLE8vLNRb0QQUw2

Sorry for the Google photos links, GitHub mobile UI doesn't seem to have an obvious image upload tool.

glance-, 2018-05-22:
I would love to see the libdivecomputer log from your device, to try to figure out whats happening here. It should be created on /sdcard/ or something alike.

@dirkhh Back to the original subject, What do you think about a checkbox in the settings to turn off all our "clever" filtering of supported devices, so if you have your own ftdi-special cable, or BT-to-serial adapter we can optionally support them. To me it feels like the semi-sane approach to resolve this sort if oddities in a sane matter.

StephenGoodall, 2018-05-22:
That log seems a bit odd......maybe I'll revert that FTDI flag and see if it changes this:

Subsurface24-g2c29cc96a34a, built with libdivecomputer v0.7.0-devel-Subsurface-NG (e97a47cca55973199715df0f818b4955e60d3a31)
INFO: Open: name=ftdi
ERROR: No such file or directory (2) [in /home/stephen/src/subsurface/libdivecomputer/src/serial_posix.c:295 (dc_serial_open)]

ecsfang, 2018-05-24:
Hi!
Just wanted to say that I have the same problem. Trying to download data from a Mares Puck Pro to a Sony XZ2, and I get the same problem as above (dc_serial_open):

```
Subsurface: v4.7.8-213-g6a91611e90cc, built with libdivecomputer v0.7.0-devel-Subsurface-NG (e97a47cca55973199715df0f818b4955e60d3a31)
INFO: Open: name=ftdi
ERROR: No such file or directory (2) [in /data/android/subsurface/libdivecomputer/src/serial_posix.c:295 (dc_serial_open)]
```

StephenGoodall, 2018-05-24:
So perhaps it's not just mine being odd...... probably just me not understanding it! I'm used to Java stack traces 😁

StephenGoodall, 2018-05-27:
Think it's failed to open the device here where name will be "ftdi":
`device->fd` = open (name, O_RDWR | O_NOCTTY | `O_NONBLOCK);`

That returns -1 which I assume is an unsuccessful attempt.

ecsfang, 2018-06-04:
This was mentioned on the forum just now:

> Like a number of people I can't download using FTDI anymore.
> I have a suunto 4i and use a OTG cable. I'm running lollipop.
> I reloaded old test versions and the last release to download correctly is 4.7.4.280 , the 4.7.5 that 
> followed and all since have failed to download.
> 

So, this "bug" seems to have been introduced with version 4.7.5.

elf128, 2018-06-24:
Hi,

I'm a bit confused by this thread.
1. While initial post was about mobile version, somewhere it gets switched to PC version. Not sure it was intentional.
2. There is no "Cressi" in the vendor's list on mobile, while I works fine on Linux and both versions use libdivecomputer, not sure why.
3. Question to @dirkhh, can you please elaborate on your statement about "certanly not adding this into official version". I do have Cressi Leonardo, why it should not be in the list? There are so many options to make adapter and access IRDA. Including DIY ( which I personally prefer ). LibDiveComputier is not using VID and PID to open device, so you're not limited by adapter's specific brand. Any chipset should work. Including various FTDI, CP21x, STM32, ATMEL... whatever, you name it. It's just a question of opening right tty.
4. I do have hardware and I would like to volunteer for testing/dev.

dirkhh, 2018-06-24:
FTDI was broken on Android - that's fixed in the latest release.
No other serial USB converter cable is supported by Subsurface-mobile on Android.
It doesn't matter if they work on Linux desktop. It doesn't matter if other software supports them on Android. They won't work with Subsurface on Android.
Adding the claim to support something that only works with a DIY cable seems wrong to me - if you want that, please build from source and install on your own device.

StephenGoodall, 2018-06-24:
Hi @dirkhh,
Just to let you know, I pulled the latest master branch, added Cressi and Leonardo to the Mobile list in core/downloadfromdcthread.cpp and built the apk.

It seems to be working now (This was the change I had previously tried) so something recent must have fixed it for this computer too. I assume this will work for the other Cressi DC's but not certain.

I have a DiveMate but I think someone else had asked about this before who had the official cable, perhaps they could try it out with that?

From what I can figure out, the DiveMate just pretends to be the official interface and presents itself with the VID and PID of the official cable.

I can send the apk file if anyone wants to give it a try, but for the more security conscious amongst you I can do a PR if required, so you can repeat it and build it yourself.

[image]

PR #1435 (2018-06-27): Add Cressi Leonardo to list of devices in Android
Adding Cressi Leonardo to the list of devices that can be selected
on Android devices.

### Describe the pull request:

- [ ] Bug fix
- [x] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:
Adding Cressi Leonardo to the list of devices selectable in Android, This now works when selecting FTDI on some devices.

### Changes made:
1) Add Cressi Leonardo to the downloadfromdcthread.cpp file

### Related issues:
Fixes #1240 

###  [... cut]
closed 2018-06-27 by dirkhh