subsurface/subsurface #1242: Can't set field "country" in dive list

sixtyfive, 2018-05-06:
After configuring File/Preferences/Georeference/Dive site layout to read Country/City/Ocean and then entering a new dive, setting location to "Atlantis/Northern Shore/Atlantic", I was expecting the "Country" field in the list of dives to read "Atlantis", but it's still not populated.

Am I simply misunderstanding how Subsurface is supposed to work here, or is this indeed a bug?

### Operating system:
GNOME 3 on top of Solus OS, all updates applied, English
Lenovo ThinkPad T430

### Subsurface version:
4.7.8
No idea about other versions...
Official release

dirkhh, 2018-05-06:
This is doing a geo reverse lookup based on the GPS coordinates of the dive site... So no, this isn't how it works.

closed 2018-05-06 by dirkhh

sixtyfive, 2018-05-06:
Would it be possible to add a section about how it _does_ work in the FAQ? The word "country" appears three times therein and nothing relates to the "Country" field in the dive list. I wasn't able to find anything in the manual, either.

dirkhh, 2018-05-06:
User manual:
13.6. Georeference
Subsurface provides a geo-lookup service (that is, given the coordinates of a dive site) derived from a click on the Dive Map panel at the bottom right of the Subsurface window, or from a GPS instrument or from the Subsurface Companion app). A search on the Internet is done to find the name of the closest known location. This function only works if Subsurface has an Internet connection. The preference of the dive site name can be configured, e.g. Country/State/City or City/State/Country (see image below).

sixtyfive, 2018-05-06:
I might have phrased misunderstandably. I did find that section of the usermanual. What I'm having trouble figuring out, though, is how to get the field marked in bright green, filled out:

[image]

mturkia, 2018-05-07:
It should come from the mentioned geo-ip lookup. If you have the coordinates, click on the button right of location. And there click on the satellite/ground station icon. This should provide you with info of country that is displayed in the divelist as well. There are some countries where the goe-ip lookup does not provide information, but this does work for me for at least Indonesia and Philippines that I just tested with.