subsurface/subsurface #1247: scripts/build.sh -mobile generates an app that cannot load.

janiversen, 2018-05-08:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When running either of 
./Subsurface/scripts/build.sh -build-deps -both
./Subsurface/scripts/build.sh -build-deps -mobile
./Subsurface/scripts/build.sh -mobile
generates a Subsurface-mobile.app, that cannot load in the simuator
failure message tells it is a faulty Info.plist.

When running packaging/ios/build.sh and Qt Creator
the generated Subsurface.app loads and works

Comparing the Subsurface.app structures, they are quite different,
scripts/build.sh includes a number of frameworks which does not seems to be
needed.

Looking at build.log it seems a bit like linking is done as a copy of the desktop app, and without the iOS SDK.

### Operating system:

macOS / iOS

mac / simulator

### Subsurface version:

Latest master

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

atdotde, 2018-05-08:
In my understanding, building with -mobile and building for the Simulator are two different things: -mobile builds the actual mobile app as (in your case) a native Mac OS app (that runs without the simulator). You can use this to develop features for the mobile app without the pain of running in the simulator or on a device (which makes for example debugging much simpler). 

the packaging/ios version builds for iOS, which can then run on a device or the simulator (which is like a device).

closed 2018-05-08 by atdotde

janiversen, 2018-05-08:
At the very least, this should be documented clearly.

But I see you closed the issue, so I guess it is OK for other to continue guessing that -mobile, does not generate anything for a mobile :-(