subsurface/subsurface #1263: Selecting EON core or EON steel on Mac does not work

janiversen, 2018-05-11:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When trying to download from a Suunto EON Core or Suunto EON Steel, there are 2 failures
a) It is not posible to select bluetooth port (you have to select "Eon").
b) It is not posible to download using the USB cable

### Operating system:

Mac High Sierra 

mac

yes both master and 4.7.8

### Subsurface version:

4.7.8 as well as master

### Steps to reproduce:

1) try to download from Suunto EON Core (or Steel)
2) /dev/tty.Bluetooth-Incoming-Port is grey and cannot be changed
3) download gives an error
4) download from Eon
5) /dev/tty.Bluetooth-Incoming-Port is dark and can be changed
6) download works

7) try to set USB port, impossible

### Current behavior:

Download only possible be selecting "Eon" with bluetooth. 

Download using Usb not possible

### Expected behavior:

"Suunto EON Core" and "Suunto EON Steel" works as "Eon" for bluetooth
Possibility to download using Usb

### Additional information:

### Mentions:

dirkhh, 2018-05-11:
Hmm. Weird.
If you can't download from USB that most likely means that you aren't building libdivecomputer against the HID library - this is a sneaky dependency that Jef isn't tracking.
The BLE thing is strange, though. EON shouldn't work AT ALL when downloading from BLE. And last time I tried this with my EON Steel on my Mac it worked (that was actually the first BLE dive computer that was supported by Subsurface...). I remember that the EON Steel was a bit tricky to get found when scanning. But I'll play with it again later today to see if I can figure out what's going on.

janiversen, 2018-05-11:
I run subsurface/scripts/build.sh -build-deps -both (or -desktop)
I expected (maybe wrongly) that to include USB etc.

I wanted to use usb on my Mac due to dm5, and the fact that my DM is paired with my iPad.

Just controlled on the iPad again (that is where my EON core is paired), selecting EON CORE and EON STEEL and BLE does not work, but selection Eon and BLE works. Nearly the opposite of your expectation. I just updated the app to latest master.

If you want logs or anything, just tell me, and I will try to provide it.

dirkhh, 2018-05-11:
libdivecomputer's autoconfigure allows one to build with and without HIDAPI. And since there are only three dive computers that require this, that seems reasonable. This is very poorly documented. You are a perfect use case of all the things that are broken with our documentation. I'll try to add a comment that explains this.

janiversen, 2018-05-11:
Would be nice if you could tell me, what do run differently :-) 

build.sh -build-deps -desktop    + whatever is missing

I have been mentoring many new people in OSS, so I am used to stumble across many irritating details, that the old people just know :-)

dirkhh, 2018-05-11:
I was hoping to have time this morning to just add that, and then things went to hell.
Get the hidapi library (I could have sworn we had this in one of the scripts, but we don't - need to add). Install it. Then run configure again for libdivecomputer - make sure it finds hidapi.
Then it should work.
I need to fix that in the script. Sorry
And I resemble the remark of being an "old people"

janiversen, 2018-05-11:
It is in build.sh !! Seems to be a question og libdivecomputer not seeing it. But please take your time, this is not a major problem.

I do not like talk of “old people” because it makes me a fossil :-) (I wrote my first program in 1975)

dirkhh, 2018-05-11:
You are the one who brought up "old people". Yet it seems that I might indeed be the youngster here, since I didn't start writing programs until '79...

dirkhh, 2018-05-12:
OK, finally had time to dig into this.
Running build.sh on Mac failed for me in the libdivecomputer build. Specifically, the tests for several compiler / language features pass, even though they should fail. This is when building on macOS 10.11 with clang-800.0.42.1 (part of Xcode 8.2.1). I have a patch that hacks around this.
Once that is fixed, libdivecomputer with a working hidapi is created and the resulting Subsurface.app happily downloads from my EON Steel over USB.
Then I went in and deleted the last two dives downloaded, deleted the pairing info on my EON Steel, and went into the download menu again. This time I clicked on "Choose Bluetooth download mode" and on the button to the right with the "...". There I clicked scan and waited for a while. The EON Steel was found. Selected it, clicked OK, started the download. Got a pairing request, accepted it, entered the six digit key, download starts and the two dives are downloaded.
Interestingly the EON Steel stays in pairing mode - and then times out with a pairing error. Even though the download worked.
So. Hmm. This seems to work exactly as intended on my Mac.
Next up my iPad Pro with the latest test app from TestFlight.
Select the EON Steel from the Connection drop down. Click download. Follow the pairing prompt. Nothing happens. Hmm. Log shows we connected fine, but then didn't get an updated value in the characteristic back. Strange.
Hit retry, it successfully downloads my dives (very, very slowly).

Summary: works for me? So let's find out why it doesn't work for you...

janiversen, 2018-05-12:
These are the steps I have just tried again:

1) Clean all files, not part of git repo, incl. all downloaded files in src.
2) in repo: git submodule init; git submodule update
3) run build.sh -build-deps -desktop
Apple LLVM version 9.1.0 (clang-902.0.39.1)
Xcode 9.3
Git is cerca 24 hours old
4) run Subsurface 4.7.8.124g2c29cc96a34a
5) Delete last 2 dives (on Suunto EON Core)
6) Connect Suunto via USB
7) Failed see screen1
8) Discoonect Suunto, to allow BT
9) try to download Suunto Core, same error
10) try to download Suunto Steel, same error
11) try to download Eon
12) Bluetooth-input-port box becomes black (enabled)
13) error see screen2
 
14) switch to iPad
15) App log (developer entry) Found new device "EON Core <serial number>" "LE:....
    Not recognized as dive computer
16) Download Suunto EON Core, using the above LE:....
17) Tries for a long time, then stops without having any dives
18) Download Suunto EON Steel, using the above LE:....
19) Tries for a long time, then stops without having any dives
20) Download Eon, using the above LE:....
21) NOT FUNNY, now this stopped working.

Having looked at the libdivecomputer sources, I can find Eon, so I think this is all some 
wrong compilation. Hopefully your patch will help both systems.

dirkhh, 2018-05-12:
I'm preparing a new TestFlight release - but I'm running out of time. Will be back in about six hours to finish. Then you should at least be able to verify this on the iPad.
For your local builds... once I've pushed my latest changes (they are still being massaged) I'd like you to try a build from Travis to see if that works for you. We need to try and figure out the differences.

janiversen, 2018-05-12:
[image]
[image]

dirkhh, 2018-05-14:
Forgot to comment here. The new TestFlight release is out and it works with my EON Steel...

janiversen, 2018-05-14:
Sorry for a stupid question, where do I doenload the testFlight for mac and iOS ?

janiversen, 2018-05-14:
My Apple ID/email is jan@casacondor.com

mfbernardes, 2018-05-14:
Install the TestFlight app.

Dirk needs to add you as a beta tester then you will receive a notification and will be able to install

Best regards,

Murillo Bernardes

dirkhh, 2018-05-14:
My mistake for not adding you - just did that. You should have received an email with the invite, @janiversen

janiversen, 2018-05-14:
No problem, just started testing.