subsurface/subsurface #1269: SAC in statistics not calculated correctly for dives with >1 gas and incomplete info

sfuchs79, 2018-05-13:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Dive with 3 different gases. No pressure info from DC. Manually entered start and end pressure for 2 out of 3 gases.
Display of SAC rate ("AMV" - sorry German) in dive info tab is correct:
[image]

Display of SAC in statistics tab is incorrect:
[image]

### Operating system:

Windows 10

### Subsurface version:

Current master 4.7.8-something

### Steps to reproduce:

Example dive:
[SAC_Example.zip](https://github.com/Subsurface-divelog/subsurface/files/1998869/SAC_Example.zip)

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

atdotde, 2018-05-14:
What is the expected behaviour? Not computing a SAC at all as information is missing?

atdotde, 2018-05-14:
Do you have gas switch events to these gases? In that case, we might be able to compute per gas SACs.

sfuchs79, 2018-05-14:
Yes, gas switch events are there. Per gas SAC is already displayed correctly in dive info tab:
[image]

In the statistics tab I would expect to see an correct average SAC based on the sections of the dive for which a per gas SAC is available. This would mean for the example dive a value between 14,0 and 16,8l/min.

sfuchs79, 2018-06-16:
This was addressed here by disabling the incorrect SAC calculation:
https://github.com/Subsurface-divelog/subsurface/pull/1289/commits/6f42a33d0ee7eed52ca2418366aeac125d289c76

I would like to keep the issue open for the moment because maybe someone finds the time to implement a CORRECT SAC calculation for the above described use case :-)