subsurface/subsurface #1275: iOS/Mac wrong number/datetime format, when using e.g. en_ES

janiversen, 2018-05-15:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I am in Spain but have all my dialogs in English, this means I have en_ES as locale (and similar on iOS).

macOS as well as iOS understands, that I expect dialogs to be English, but number formatting and date time formal to follow Spanish (European) rules.

Subsurface looks at the "en_", determines I am an American and decides to give me dialogs in English (nice) and number/datetime format as US ( 1,000.00 instead of 1.000,00, and 5/15/18 instead of 15/5/18).

When I set locale to es_ES Subsurface works as expected.

This might be a combination of a Subsurface and Qt problem. The solution used in some other
projects is to (in the program) change e.g. en_ES -> es_ES while setting up Qt, but requesting all text to be in "en".

### Operating system:

High Sierra, and iOS 11.3

### Subsurface version:

Releases + testFlight + master

### Steps to reproduce:

1) set local to en_ES
2) Start subsurface, watch numbers
3) set local to es_ES
4) restart Subsurface, watch numbers

### Current behavior:

US number/datetime formatting

### Expected behavior:

number/datetime formatting according to _<country> in locale

### Additional information:

### Mentions:

neolit123, 2018-05-15:
hi, could you provide a link for this:
> This might be a combination of a Subsurface and Qt problem. The solution used in some other
projects is to (in the program) change e.g. en_ES -> es_ES while setting up Qt, but requesting all text to be in "en".

janiversen, 2018-05-28:
switched to es_ES.

closed 2018-05-28 by janiversen