subsurface/subsurface #1315: Building against Qt-5.11.0 fails

vesath, 2018-05-23:
Building git master against Qt-5.11.0 goes as far as:

```
[ 77%] Linking CXX executable TestPreferences
[ 78%] Building CXX object desktop-widgets/CMakeFiles/subsurface_interface.dir/divecomputermanagementdialog.cpp.o
[ 78%] Built target TestPreferences
[ 78%] Building CXX object desktop-widgets/CMakeFiles/subsurface_interface.dir/divelistview.cpp.o
[ 78%] Linking CXX executable TestTagList
[ 78%] Building CXX object desktop-widgets/CMakeFiles/subsurface_interface.dir/diveplanner.cpp.o
[ 78%] Built target TestTagList
[ 79%] Building CXX object desktop-widgets/CMakeFiles/subsurface_interface.dir/diveshareexportdialog.cpp.o
[ 79%] Building CXX object desktop-widgets/CMakeFiles/subsurface_interface.dir/downloadfromdivecomputer.cpp.o
/build/src/subsurface/desktop-widgets/divelistview.cpp: In constructor 'DiveListView::DiveListView(QWidget*)':
/build/src/subsurface/desktop-widgets/divelistview.cpp:44:10: error: invalid use of incomplete type 'class QHeaderView'
  header()->setContextMenuPolicy(Qt::ActionsContextMenu);
          ^~
In file included from /usr/include/qt/QtWidgets/QTreeView:1,
                 from /build/src/subsurface/./desktop-widgets/divelistview.h:15,
                 from /build/src/subsurface/desktop-widgets/divelistview.cpp:23:
/usr/include/qt/QtWidgets/qtreeview.h:53:7: note: forward declaration of 'class QHeaderView'
 class QHeaderView;
       ^~~~~~~~~~~
/build/src/subsurface/desktop-widgets/divelistview.cpp:100:10: error: invalid use of incomplete type 'class QHeaderView'
  header()->setStretchLastSection(true);
          ^~
In file included from /usr/include/qt/QtWidgets/QTreeView:1,
                 from /build/src/subsurface/./desktop-widgets/divelistview.h:15,
                 from /build/src/subsurface/desktop-widgets/divelistview.cpp:23:
/usr/include/qt/QtWidgets/qtreeview.h:53:7: note: forward declaration of 'class QHeaderView'
 class QHeaderView;
       ^~~~~~~~~~~
/build/src/subsurface/desktop-widgets/divelistview.cpp: In member function 'void DiveListView::reload(DiveTripModel::Layout, bool)':
/build/src/subsurface/desktop-widgets/divelistview.cpp:422:10: error: invalid use of incomplete type 'class QHeaderView'
  header()->setSectionsClickable(true);
          ^~
In file included from /usr/include/qt/QtWidgets/QTreeView:1,
                 from /build/src/subsurface/./desktop-widgets/divelistview.h:15,
                 from /build/src/subsurface/desktop-widgets/divelistview.cpp:23:
/usr/include/qt/QtWidgets/qtreeview.h:53:7: note: forward declaration of 'class QHeaderView'
 class QHeaderView;
       ^~~~~~~~~~~
/build/src/subsurface/desktop-widgets/divelistview.cpp:423:101: error: no matching function for call to 'DiveListView::connect(QHeaderView*, const char [21], DiveListView*, const char [20], Qt::ConnectionType)'
  connect(header(), SIGNAL(sectionPressed(int)), this, SLOT(headerClicked(int)), Qt::UniqueConnection);
                                                                                                     ^
[...]
```
For more information see [the full build log](https://github.com/Subsurface-divelog/subsurface/files/2029260/build.log).

I'm running Arch Linux with the [testing] repository enabled which currently ships Qt-5.11.0 packages. My build environment is otherwise the same as usual.

vesath, 2018-05-23:
These errors are fixed by pull request #1316 . There's a segfault with list sorting that remains to be fixed for subsurface to be fully functional with Qt-5.11, though.

dirkhh, 2018-06-25:
Qt 5.11.1 fixes the remaining issues.

closed 2018-06-25 by dirkhh

navneetccna, 2018-08-28:
getting same issue in Qt 5.11.1

error: invalid use of incomplete type ‘class QHeaderView’
     mTableWidget->getTableView()->horizontalHeader()->setStretchLastSection(true);
                                                     ^~

dirkhh, 2018-08-28:
did you do a `make clean`? I build current master against 5.11.1 on several platforms (including macOS and two Linux flavors) without any problems...