subsurface/subsurface #1326: iphone start time, bad user experience.

janiversen, 2018-05-25:
### Describe the issue:

- [ ] Bug
- [ x] Change request
- [ ] New feature request
- [x ] Discussion request

### Issue long description:

when ssrf is started from cold (not loaded in memory, but have been started
several times before), it displayed the splash screen for 10sec, and 12sec if
bluetooth is disabled.

Having done a bit of debugging, there are strange things going on, e.g.
plotDive is called twice (even though the divelist is displayed without a
profile).

### Operating system:

iOS

iPhone 6s

### Subsurface version:

master and testFlight

### Steps to reproduce:

1) remove ssrf from memory
2) start ssrf

### Current behavior:

far too slow

### Expected behavior:

startup tim of 1-2 seconds.

the main problem is in our constructors, which not only construct
the object but also does handling. A better way, would be to have
a near empty constructor, and a init(), which as a sideeffect would allow
the sequence to be controlled.

in order to catch conflict the instance() could check if init was called and e.g.
do an assert()

### Additional information:

### Mentions:

janiversen, 2018-05-25:
remember iOS does not have JIT, so precompiling the qml files might save time.

bstoeger, 2018-05-27:
I generally agree with the notion of simplifying / decluttering constructors and streamlining the start-up sequence. This is a mess.

But I disagree with the constructor/init() split. We have that in a few special cases and this makes things just even more convoluted.

> in order to catch conflict the instance() could check if init was called and e.g.
do an assert()

With a few exceptions (e.g. MainWindow), the whole point of the `instance()` methods is that the objects are constructed as late as possible (when they are needed). I don't see how calling the constructor *and* `init()` before `instance()` will make things faster.

I think what we need is an analysis why certain function are called multiple times during startup and which functions could be delayed to e.g. when a component is actually shown.

janiversen, 2018-05-27:
Sorry I might have analyzed things wrong, or extrapolated wrongly from what I  dlievd to have seen. It also seems I explained my idea badly.

I see a lot of global objects being created during startup, because they are global, and I thought the static instance method was used to reference these global objects. Presuming I am right (for the majority) then 
1) moving the content of the constructor to a init(), apart from variable initialization
2) in instance() check if init() was called and if not call it before returning.

would delay a lot of what happens during startup until the objects are actually used, and as a sideffect the sequence would be controlled and not random as are the case of global variables.

I totally agree we need to analyze why e.g. plotdive is called twice, I am doing that slowly,  every time I start a test.

I hope my presumption is correct, if not, my only excuse is that I am still very new to the source.

bstoeger, 2018-05-27:
Could it be that you are not aware of the fact that function-local statics are only constructed when the function is entered the first time? Any modern compiler guarantees that this is thread safe.

Consider for example:
```
GasSelectionModel *GasSelectionModel::instance() 
{ 
        static GasSelectionModel self; 
        return &self; 
} 
```
The `self` object is constructed the first time the `instance()` function is called.

Some time ago I wrote a patch that unified all the `instance` functions to use this scheme. I never posted a PR, because in some cases the global objects need a parent object. These objects should be "unglobalized", but so far I was too lazy to do that. But perhaps I should pick this up again one day.

janiversen, 2018-05-28:
I do know about this way of doing things, which are rather elegant but Ups to be honest, I looked at the link map output to find global objects, but did not check if they were weak.

I have filed PR #1337 to get a more consistent analysis of where the problems are, and how different they are on different devices.

janiversen, 2018-05-28:
Updated the issue to reflect a user view, allowing me to forward my concern, and let others do the thinking.

#1337 is closed, because the demands outweighed the benefits for something that is temporary.

neolit123, 2018-05-28:
comments on the actual bottle neck:
https://github.com/Subsurface-divelog/subsurface/pull/1337#issuecomment-392608673

janiversen, 2018-05-28:
Just read the comment and add to that that JIT is not available on iOS. This mix explains a lot of the strange things I see.

But all in all if 5 sec is quite acceptable, then its better to close this issue

closed 2018-05-28 by janiversen

dirkhh, 2018-05-28:
I disagree. I think this is definitely worth investigating - I really want to have better data on where we spend the time.

@janiversen  don't give up so quickly, please

janiversen, 2018-05-29:
@dirkhh is was not quickly, it was a simple choice of either continue fighting against a core developer who clearly see the work as bad, or move on.

I commented more in detail on the closed PR, sorry to disappoint you.