subsurface/subsurface #1331: mobile: black subsurface logo due to Kirigami bug

mfbernardes, 2018-05-26:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Since the last Kirigami update the subsurface logo is black, as below:

 [image]

This is due to Kirigami having changed the Icon implementation, and erroneously tinting icons that should not be tinted.

I'll report and track with Kirigami dev.

### Operating system:

iOS 11.3

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@notmart

mfbernardes, 2018-05-26:
I've split #1267 because I found the cause of different icons missing. This one is unrelated to the others.

Some tests reported on #1307

neolit123, 2018-05-26:
@janiversen @mfbernardes 
please, close this issue if it's a duplicate or if we have a fix in `master` already.
thanks.

janiversen, 2018-05-26:
@neolit123 I just tested with master on my iPhone, problem is still there, just as @mfbernardes describes it. The idea it is being tinted, sounds very logical from the debugging I have done (the image is loaded from disk/memory, but then changed on its way).

commit f9b050a2 (2018-06-16): QML UI: build our own icon and header for global drawer

Kirigami appears to have a bug that makes it fail to show our icon.
With this we can be much more flexible in what we show in the top area
of the global drawer.

Fixes #1331
commit d5969960 (2018-06-16): QML UI: build our own icon and header for global drawer

Kirigami appears to have a bug that makes it fail to show our icon.
With this we can be much more flexible in what we show in the top area
of the global drawer.

Fixes #1331
commit c8ef3d79 (2018-06-17): QML UI: build our own icon and header for global drawer

Kirigami appears to have a bug that makes it fail to show our icon.
With this we can be much more flexible in what we show in the top area
of the global drawer.

Fixes #1331
closed 2018-06-17 by dirkhh