subsurface/subsurface #1344: Request: Add Shearwater Teric

Ghalr, 2018-05-31:
Please
Do you plan to add Shearwater: Teric support ?
https://www.shearwater.com/products/teric/

Thanks.

### Describe the issue:
- [ ] Bug
- [ ] Change request
- [X] New feature request
- [ ] Discussion request

dirkhh, 2018-06-18:
That's a nice looking dive watch. I haven't seen one, yet, but I reached out to Shearwater to see if I can find out more about it.
Have you tried to connect to it over BLE? If you use something like nRF Connect on an Android phone, can you connect & BOND? What name does it show up under? What characteristics does it offer?
Assuming it's fairly similar to existing Shearwater models, we may be able to add support fairly easily...

dirkhh, 2018-06-20:
After talking to Shearwater I learned that this dive computer uses a new dive log format that hasn't been publicly documented. Obviously we could try to reverse engineer this, given access to a Teric plus sufficient time and motivation... or we can just wait until Shearwater makes this information available to us.

Ghalr, 2018-06-20:
Apologize I do not have this computer but as a Shearwater user I might buy it a bit later, or next year. But I think it more than welcome to have such a powerfull computer into your compatible devices.

Maybe some users have it and can provide you informations you requested ?

az-chris, 2018-07-06:
I just got one and will be diving it this weekend, want me to upload or send my dive log somewhere?

dirkhh, 2018-07-06:
The challenge is that it uses a new format. Shearwater gave me access to one as well, I will see how different the data are.
We are certainly always interested in test data.

stefanmb, 2018-07-13:
I can also provide logs from my Teric if given instructions on how to do so. It would be great to have native Subsurface support!

dirkhh, 2018-07-13:
Thanks, appreciate all the offers. Support will come, I am waiting to hear from Shearwater - they promised an answer "soon"...

lusitania, 2019-05-01:
Support for the Teric was added in [4.8.2](https://subsurface-divelog.org/2018/09/announcing-subsurface-4-8-2/). Close?

closed 2019-05-01 by mturkia