subsurface/subsurface #1346: Syncing with cloud/ Creating commit fails

exxus, 2018-06-01:
### Describe the issue:
- [x] Bug

### Issue long description:
Subsurface fails to sync with cloud storage. A red banner shows "creating commit failed".

### Operating system:
archlinux

### Subsurface version:
Self compiled binary:
Subsurface v4.7.8-289-g9611e92cf0d9,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (e97a47cca55973199715df0f818b4955e60d3a31)
built with Qt Version 5.11.0, runtime from Qt Version 5.11.0
built with libgit2 0.27.0

Also happens with Appimage:
Subsurface v4.7.8,
built with libdivecomputer v0.7.0-devel-Subsurface-branch (d02f1c3cdcc6a04d085538578d872ec6e3282382)
built with Qt Version 5.9.3, runtime from Qt Version 5.9.3
built with libgit2 0.26.0

Also happens with Archlinux package:
Subsurface v,
built with libdivecomputer v0.7.0-devel-Subsurface-branch ()
built with Qt Version 5.11.0, runtime from Qt Version 5.11.0
built with libgit2 0.27.0

### Steps to reproduce:

1) open local logbook *.ssrf
2) select file-> save to cloud storage
3) red banner appears, "creating commit failed", new dives are not saved to cloud

### Current behavior:
 No new commit  is created in ./subsurface/cloudstorage/hash/

### Expected behavior:
Create a new commit with the changes from the local file and push it to the cloud server.

### Additional information:
 Command line log:
...
cloud URL set as "https://cloud.subsurface-divelog.org//git/___@gmail.com[___@gmail.com]"
Saving cloud storage to: "https://cloud.subsurface-divelog.org//git/___@gmail.com[__@gmail.com]"
git storage: Synchronising data file
git storage: Sync with cloud storage
git storage: Successful cloud connection, fetch remote
git storage: Done syncing with cloud storage
git storage: Preparing to save data
git storage: Start saving data
git storage: Start saving dives
git storage: Done creating local cache

[image]

exxus, 2018-06-01:
After doing a debug build with optimization turned (-O0) and running subsurface with gdb to find the root of the problem, the synchronization worked (git commit was created).

neolit123, 2018-06-01:
@exxus

something not working with a release build and working with the debug build is not a good sign.
it could be due to memory corruption.

a couple of things:
1. i've tried the steps on our latest dev. version on Windows and i cannot reproduce this.
could you confirm this with our latest dev. build:
https://github.com/Subsurface-divelog/subsurface/releases/tag/untagged-7d14aac1e4451528b2c4

2. (if 1. doesn't work) have you tried clearing the local and remote storage before trying the steps again?
for that you would need to clear the local commit history and force push that to the remote.

exxus, 2018-06-05:
@neolit123 

1. the link you provided seems to be broken, but with this build: https://github.com/Subsurface-divelog/subsurface/releases/tag/continuous (1c2ed7f)
I still can't sync with the cloud service (creating commit failed)

2. what do you mean by "clear local commit history"? Should I create a new, empty repositroy and force push that to the remote?

3. running with -O2 and debug symbols in gdb, I see the  following line is executed:
https://github.com/Subsurface-divelog/subsurface/blob/2209054226f463a3fd0b88a34e9dcd18e15b363b/core/save-git.c#L1148
and than fails with the creating commit failed message
https://github.com/Subsurface-divelog/subsurface/blob/2209054226f463a3fd0b88a34e9dcd18e15b363b/core/save-git.c#L1272

running -O1 build works.
As you mentioned memory corruption, running through valgrind gives the following summary:
```
==17210== LEAK SUMMARY:
==17210==    definitely lost: 196,665 bytes in 546 blocks
==17210==    indirectly lost: 82,134 bytes in 1,244 blocks
==17210==      possibly lost: 12,036 bytes in 134 blocks
==17210==    still reachable: 4,697,135 bytes in 41,304 blocks
==17210==                       of which reachable via heuristic:
==17210==                         length64           : 3,192 bytes in 57 blocks
==17210==                         newarray           : 2,088 bytes in 49 blocks
==17210==         suppressed: 0 bytes in 0 blocks
==17210== Reachable blocks (those to which a pointer was found) are not shown.
==17210== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==17210== 
==17210== For counts of detected and suppressed errors, rerun with: -v
==17210== Use --track-origins=yes to see where uninitialised values come from
==17210== ERROR SUMMARY: 67532 errors from 372 contexts (suppressed: 0 from 0)
valgrind --leak-check=yes ./subsurface  38,18s user 0,46s system 40% cpu 1:36,37 total
```
As a workaround, since I can build my own binary, I just turn off optimization and I can sync to cloud.

dirkhh, 2018-06-25:
I am running on Arch myself and have never seen this issue - admittedly I build from source and don't use the Arch Subsurface package, but @exxus said he tried building from source as well.

neolit123, 2018-07-06:
@exxus 
given you are building from source, perhaps you can help us debug this?

1) open `core/save-git.c`
2) locate `static int get_authorship(git_repository *repo, git_signature **authorp)`
3) in this function modify this line:
```
return git_signature_now(authorp, user.name, user.email);
```
to this:
```
int result = git_signature_now(authorp, user.name, user.email);
printf("-----> %d, %p, %s, %s\n", result, authorp, user.name, user.email);
return result;
```

4) reproduce the issues steps
5) share the output of the `----> .... ` line

thanks.

dirkhh, 2018-07-06:
```
==6163== Invalid read of size 1
==6163==    at 0xA64E060: __strchr_avx2 (in /usr/lib/libc-2.27.so)
==6163==    by 0x4EE35D0: git_signature_new (in /usr/lib/libgit2.so.0.27.2)
==6163==    by 0x4EE38DB: git_signature_now (in /usr/lib/libgit2.so.0.27.2)
==6163==    by 0x3F069E: get_authorship (save-git.c:1066)
```
That's code that hasn't changed in ages - but of course you are using a newer `libgit2` than most of our builds. Anyway, it's not what's causing the crash - Valgrind simply stopped reporting errors after 100 of them (the libraries we use contain a lot of false positives with Valgrind).

exxus, 2018-07-06:
@neolit123 

`-----> -1, 0x7fffd4ef5438, Subsurface, `

neolit123, 2018-07-06:
thank you @exxus .
`-1` is an error code.
`0x7fffd4ef5438` looks like a valid pointer.
`Subsurface` is  valid name, but the email is empty.

1) please open `core/linux.c`
2) navigate to `subsurface_user_info()`
3) modify the function to look like this and reproduce the steps:

```
void subsurface_user_info(struct user_info *user)
{
	struct passwd *pwd = getpwuid(getuid());
	const char *username = getenv("USER");

	if (pwd) {
		if (!empty_string(pwd->pw_gecos))
			user->name = pwd->pw_gecos;
		if (!username)
			username = pwd->pw_name;
	}
	printf("-----> %p, %s, %s\n", pwd, username, user->name);
	if (!empty_string(username)) {
		char hostname[64];
		struct membuffer mb = {};
		gethostname(hostname, sizeof(hostname));
		printf("-----> %s\n", hostname);
		put_format(&mb, "%s@%s", username, hostname);
		user->email = mb_cstring(&mb);
		printf("-----> %s\n", user->email);
		free_buffer(&mb);
	}
}
```
4) share the `---->` output lines.

exxus, 2018-07-06:
```
-----> 0x7f0274bccd80, exus, Subsurface
-----> exzen
-----> exus@exzen
-----> -1, 0x7ffda539b798, Subsurface, 
```

neolit123, 2018-07-06:
@exxus thank you
than's a bug introduced here: 
https://github.com/Subsurface-divelog/subsurface/commit/b5ce06c8b958cdc51761c31697189db81baf4423

```
user->email = mb_cstring(&mb);
free_buffer(&mb);
```

we have to fix it for 4.8.1.

exxus, 2018-07-06:
I see, the memory is freed, so user->email now points to an invalid memory location.
Removing line 
`free_buffer(&mb);`
solves the problem but introduces a memory leak.

neolit123, 2018-07-06:
yes, i will submit a fix for this later today or doing the weekend.

closed 2018-07-08 by dirkhh