subsurface/subsurface #1351: Drawing of the computer-reported ceiling doesn't "end" when deco ceiling is cleared

gextyr, 2018-06-03:
### Describe the issue:
- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Drawing (on chart) of the computer-reported ceiling doesn't seem to end, at least for Ratio iX3M Tech+, after deco/ceiling has cleared.
[image]
[image]

Attached are screen shots of Subsurface compared to Ratio's (horrible) DiveLogger software.

### Operating system:
Windows 10 Pro (English) x64 Build 17134
Desktop - Intel Core i7-2600K 3.4Ghz & 32GB
Haven't tried on other devices/OSes yet, but I can/will if requested.

### Subsurface version:
4.7.8 (official release from subsurface-divelog.org)
Haven't tried on other version.

### Steps to reproduce:
Imported dive log (from computer) and toggled computer-reported ceiling

### Current behavior:
Computer-reported ceiling/deco doesn't show as "cleared" on graph, even after computer showed clear (and Ratio DiveLogger shows clear.)

### Expected behavior:
Expect the computer-reported deco to match DiveLogger.

### Additional information:
Included relevant dive from log.
[gextyr.zip](https://github.com/Subsurface-divelog/subsurface/files/2066430/gextyr.zip)

Note - given the data, it seems that (due to ocean swell and current) I didn't hold at exactly 20ft/6m, which causes numerous recalculations of stop time.  However, the computer and DiveLogger both behave as expected.

janmulder, 2018-06-12:
A little analysis. I checked a dive from my OSTC with reported ceilings, and that works just fine. Comparing on XML level explains a difference. My OSTC profile has a sample `in_deco='0' stopdepth='0.0 m'` that marks the end of the reported ceiling. The XML provided here shows only `stoptime='4:00 min' stopdepth='6.0 m'` and there is no sample showing the end.

This could be an issue in the libdivecomputer parser, and to pursue that further we would need a libdivecomputer dump file (see Import from dive computer screen), in order to simulate the import from a real device.

gextyr, 2018-06-12:
@janmulder - Right now, when I select "Save livdivecomputer dump file", I get the an error dialogue: "Dive data import error".  (The dives import fine with that de-selected.)  I will look into it when I have the time... but do you have any ideas as to why that might happen?  I am running in a Win10 environment, with the official Windows build - not sure if that matters.  I will spin up a WSL or Linux system to test later.
--
I also noticed in the DiveLogger software log files, there are "firstStopDepth" (decimeters) and "firstStopTime" (seconds) xml nodes, that change from >0 to =0 as deco is cleared.  So it seems you are correct that there is a disconnect during the parsing/translation of the computers data.

janmulder, 2018-06-12:
Guessing, but this cloud be one of those cases that require a command-line based download of raw dive data from the device using dctool. Available here: http://libdivecomputer.org/builds/stable/windows/dctool.exe with a command like:

`dctool.exe -v -f <family name of Ratio iX3M Tech+> download -o dive.%n.bin -f raw <COMx>`

I do not know the family name to be used. @jefdriesen?

gextyr, 2018-06-12:
Ah, thanks!  The Ratio branded devices are all in the "idive" family.  The model for the iX3M Tech+ is 0x45 (per libdc/src/descriptor.c)  I included the dump, as well as Ratio's DiveLogger xml log for comparison.

dctool.exe -v -f idive -m 0x45 download -o dive.%n.bin -f raw COM5

[dive.0010.zip](https://github.com/Subsurface-divelog/subsurface/files/2094796/dive.0010.zip)

janmulder, 2018-06-12:
@jefdriesen: I see "Unexpected packet length" errors when trying to import the above provided .bin via de simulator. Highly likely because the simulator is built against an older libdivecomputer (before that packet length change from 0x36 bytes to 0xC0 bytes). Is there a way to build the simulator to a more recent version of libdivecomputer?

jefdriesen, 2018-06-12:
You need to pass the correct model number with the -m 0x45 option. And then it should work.

The simulator doesn't need a recent libdivecomputer version. I'm pretty sure it won't even build anymore against recent versions. Someday I need to take the time to integrate it properly in the libdivecomputer code!

janmulder, 2018-06-12:
Unfortunately it does not work. The simulator provides some samples, and then crashes hard with a SIGSEGV.

@gextyr: But I'm confused here. Are we talking about a `{"Ratio",      "iX3M Tech+",    DC_FAMILY_DIVESYSTEM_IDIVE, 0x24, DC_TRANSPORT_SERIAL, NULL}` or a `{"Ratio",      "iDive Tech+",   DC_FAMILY_DIVESYSTEM_IDIVE, 0x45, DC_TRANSPORT_SERIAL, NULL}`?

gextyr, 2018-06-12:
@janmulder - ah yes, perhaps there is some confusion between these.  
I think 0x24 is the correct one - I guess I just missed that before.  DiveSystem/Ratio's naming convention for their computers is horrible.  I attached a dump using that parameter in case there is a difference in packet length between the two models.

[dive.0010.zip](https://github.com/Subsurface-divelog/subsurface/files/2095397/dive.0010.zip)

janmulder, 2018-06-12:
About to give-up. The 2 .bin files are identical, so that does not help (unfortunately). Tried the simulator with both -m 0x24 and 0x45, and even tried to import as Divesystem, but no success. Fixing this without test-platform seems a bit risky.

jefdriesen, 2018-06-13:
The reason why it doesn't work is because the *.bin.header file is missing. This contains some auxiliary data like that model number, firmware version, etc that is not present in the main memory dump file. In this case, passing the model number with the -m option isn't enough because the data format is firmware dependent.

You can create the 26 byte header file from the data in the logfile: 

`grep 'Event: vendor=' $LOGFILE | sed 's/^.*vendor=\([0-9a-fA-F]*\).*$/\1/' | head -n 1 | xxd -p -r > $HEADERFILE`

But if you just want to parse the dive files, then you can use:

`./dctool -f idive -m 0x24 parse -o dctool.xml dive.*.bin`

The naming of the different models is indeed a bit confusing. But what matters for libdivecomputer is whether the model number is  >= 0x22 (e.g. the newer Ratio models) or not (the older DiveSystem models). So as long as you pass a number in the correct range, it will just work.

gextyr, 2018-06-13:
@janmulder - The attached now contains the header file (as @jefdriesen instructed) and the dctool parse output.  I have also included 12 dives, instead of just the 1.  Only a few of them have computer-reported ceiling/deco - look at dives numbers 2, 3, and 5.  They all exhibit the same behavior (ceiling not clearing on graph).

I am going through this as well... however, I wanted to leave more data here in case you find it useful.

[dives.tar.gz](https://github.com/Subsurface-divelog/subsurface/files/2098764/dives.tar.gz)

janmulder, 2018-06-13:
@gextyr and @jefdriesen : yes, with this I can reproduce the problem. Thanks for the input, and will try to understand the root cause of the issue.

janmulder, 2018-06-14:
@jefdriesen. I believe I found the issue in libdivecomputer. But lacking interface specs of that device, I leave it to you to evaluate it further.

```
diff --git a/src/divesystem_idive_parser.c b/src/divesystem_idive_parser.c
index 413d7c4..bbd72d4 100644
--- a/src/divesystem_idive_parser.c
+++ b/src/divesystem_idive_parser.c
@@ -377,7 +377,7 @@ divesystem_idive_parser_samples_foreach (dc_parser_t *abstract, dc_sample_callba
                        decostop = array_uint16_le (data + offset + 21);
                        decotime = array_uint16_le (data + offset + 23);
                        tts      = array_uint16_le (data + offset + 25);
-                       if (tts == 0x7FFF) {
+                        if (tts == 0xFFFF) {
                                tts = INVALID;
                        }
                } else {
```

jefdriesen, 2018-06-14:
Hmm, I'll need to look into this again. The documentation says 0xFFFF, but when I implemented this code, the data I had available contained 0x7FFF. Hence the difference. See commit 4ccfa51faf8c01ed1ca9d096fba5f49f030f3e43 for the details. Maybe this is something that changed with a firmware update?

cyberdiamond, 2018-07-23:
Jef I am not sure if this changed or if it was documented wrong but I can confirm this behaviour in all versions of firmware in the last 2 years.

jefdriesen, 2019-02-22:
According to Ratio, there was indeed a mistake in their documentation. The correct value for infinite tts is indeed 0x7FFF.

Now, the problem appears to be that once the deco stops are cleared, the dive computer switches back to ndl mode with an infinite tts. Because libdivecomputer doesn't have a way to represent "infinite", those sample values are simply not reported (e.g. ndl times are only reported when they have a finite time). I suspect subsurface incorrectly takes the value from the previous sample. Thus, if there is no deco/ndl sample reported anymore, the deco stop gets extended by subsurface all the way to the end of the dive.

janmulder, 2019-02-24:
@jefdriesen: do I understand correctly that the idive parser does not report any change to signal the end of the deco mode. If that is true, how would Subsurface have to derive that endpoint? So, yes, its very likely that we take data from the previous sample, simply as there is no new data.

jefdriesen, 2019-02-28:
The fact that you don't get another deco sample indicates the end.

Stefano1969, 2019-03-13:
I used the same way for an another application: 
when firstStopDepth  became = 0, the deco is ended, and so the ceiling/deco could be show  as "cleared" .

Following an extract of the file XML, produced by the Ratio DiveLogger:

[image]

cyberdiamond, 2019-06-24:
Any news on getting this fixed?

jefdriesen, 2020-02-28:
Fixed in libdivecomputer commit 8a1d32d319eaa62a550c07349ef16f8dcb923931.

jb2cool, 2020-03-08:
@gextyr @cyberdiamond if you would like to try a test build where this should be fixed, there is one available here https://github.com/Subsurface-divelog/subsurface/releases/tag/ci-release

mikeller, 2026-10-17:
Hi @gextyr,

Thanks for the months of patient testing and data on this one. jefdriesen confirmed: "Fixed in libdivecomputer commit 8a1d32d319eaa62a550c07349ef16f8dcb923931", addressing the root cause you and the others in this thread helped pin down, the idive-family parser not recognising the absence of further deco samples as the ceiling actually clearing.

Closing this as fixed. Please reopen if your iX3M Tech+ (or another idive-family computer) still shows a ceiling that never clears on a current build.

closed 2026-10-17 by mikeller