subsurface/subsurface #1356: List of dives disappears after clicking table header to sort

parsantos, 2018-06-04:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

After opening SS, and loading dives from cloud storage, I click on the "Location" column header, and the list of dives disappears.
After reloading the dives, the list is shown again.

### Operating system:

Arch Linux updated with latest releases

### Subsurface version:

Subsurface v4.7.8-296-g9293268027e1,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (e97a47cca55973199715df0f818b4955e60d3a31)
built with Qt Version 5.11.0, runtime from Qt Version 5.11.0
built with libgit2 0.27.0

### Steps to reproduce:

1) File -> Open Cloud Storage
2) Click any column header in Dive List

### Current behavior:

Table with dives disappears

### Expected behavior:

Table is sorted by location

### Additional information:

Log:
QPainter::begin: Paint device returned engine == 0, type: 2
QPainter::setFont: Painter not active
QPainter::setOpacity: Painter not active
QPainter::end: Painter not active, aborted

### Mentions:

neolit123, 2018-06-05:
thank you for the report, @parsantos.

i cannot reproduce this bug....strange.
i've seen the error itself before - it happens with a paint target is not initialized properly.

some questions:
1) are you using this AppImage `Subsurface-4.7.8-296-g9293268027e1-x86_64.AppImage` from here:
https://github.com/Subsurface-divelog/subsurface/releases/tag/continuous
2) can you reproduce the bug if you start a new log book and add some dives to it, instead of loading from the cloud?

parsantos, 2018-06-05:
Hi,

1. I've built this from source:
git checkout master
git pull

2. if I start a new log book the behaviour is the same: same QPainter errors and dive list disappears

neolit123, 2018-06-05:
@parsantos does the same happen with the AppImage mentioned above?

parsantos, 2018-06-06:
Hi @neolit123,

I just tried the AppImage. It works ...

neolit123, 2018-06-06:
@parsantos thanks for confirming,

well...there are so many things that can go wrong in a build, so it's really hard to tell what's causing this.
i'm going to have to mark this as `bug-not-in-subsurface`, but please keep me up to date about your findings and let me know how can i help further.

maybe try a couple of things:
- different compiler
- different Qt version

parsantos, 2018-06-07:
Mersi @neolit123 :)

I'll keep you posted on any findings

dirkhh, 2018-06-18:
We don't officially support Qt 5.11, yet (because of a couple of other bugs that the Qt developers pointed out to us). Right now we recommend building against Qt 5.9.5 or Qt 5.10.1

p-neves, 2018-06-18:
Hi all
I see this happening also on my builds. 
Steps to reproduce:
1) Open Subsurface
2) Select one dive
3) Edit something (e.g. Notes)
4) Apply changes

Divelist disappears. This does not happen with Appimage.
I've also compiled with QT 5.11 on arch...

parsantos, 2018-06-21:
After yesterday's Qt5 update to 5.11.1, I have to say that the issue is no longer there, and this can be closed.

Thanks everyone!

Versions details:
Subsurface v4.7.8-394-g74d1afc0d5cd,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (e97a47cca55973199715df0f818b4955e60d3a31)
built with Qt Version 5.11.1, runtime from Qt Version 5.11.1
built with libgit2 0.27.0

closed 2018-06-21 by atdotde