subsurface/subsurface #1372: kirigami: Binding loop error on mobile-desktop Mac

janiversen, 2018-06-10:
### Describe the issue:

- [x? ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

During some extensive testing with mobile on my Mac I noted, what I thought was a new error:

INFO: qrc:/org/kde/kirigami/templates/private/BackButton.qml:25:1: QML ToolButton: Binding loop detected for property "enabled"

I tried to bisect it, and found it is in 2209054226f463a3fd0b88a34e9dcd18e15b363b from June 1.

Funny thing is that I do not see the same error on my iPad (of course building the same version).

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

neolit123, 2018-06-10:
> I tried to bisect it, and found it is in 2209054 from June 1.

that commit is a documentation change, so it couldn't possibly cause a:

`INFO: qrc:/org/kde/kirigami/templates/private/BackButton.qml:25:1: QML ToolButton: Binding loop detected for property "enabled"`

janiversen, 2018-06-10:
Sorry for my english, I meant the error is still present in that commit, so the problem seems to be introduced earlier.

janiversen, 2018-06-11:
tested a bit further today

the error is present in commit
f47f2773fd4de5bfbb8d90e5e00dc18e0c18e5bb
from may 19, so the problem is before this commit

When I tried to test
1ed8c87d6394bec8d115d1fd90bdb08e207d20eb
from may 16, I got a lot of error messages:
qrc:///qml/main.qml:5 module "QtQuick.Window" is not installed
qrc:///qml/main.qml:7 module "QtQuick.Layouts" is not installed
qrc:///qml/main.qml:4 module "QtQuick.Controls.Material" is not installed
qrc:///qml/main.qml:6 module "QtQuick.Dialogs" is not installed
qrc:///qml/main.qml:8 module "QtQuick.Window" is not installed
qrc:///qml/main.qml:3 module "QtQuick.Controls" is not installed
qrc:///qml/main.qml:5 module "QtQuick.Window" is not installed
qrc:///qml/main.qml:7 module "QtQuick.Layouts" is not installed
So I do not know if the problem exist in that commit.

Any idea what changed between may 16 and may 19, that could cause QtQuick.Window not to be installed ?

janiversen, 2018-06-12:
tag v4.7.8 works and the error is not present.

janiversen, 2018-06-16:
It is fixed on master, but I was doing bisect between master and 4.7.8 so I probably tested before your commit.

I alway run "LIBRARY_PATH=.... make" and then "LIBRARY_PATH=.... make install". Very easy now with the new makefile :-)

dirkhh, 2018-06-18:
This is the offending line:
```
    enabled: !Settings.isMobile && (__appWindow.pageStack.currentIndex > 0 ||
                       applicationWindow().pageStack.contentItem.contentX > 0)
```
I'm a bit puzzled how this could be recursive - none of these should change depending on whether the BackButton is enabled.

janiversen, 2018-06-18:
do you mind giving me a file reference ?

janiversen, 2018-06-18:
got it: kirigami/src/controls/templates/private/BackButton.qml 

That is not recursive, but called multiple times, at least once for every of our pages.

janiversen, 2018-09-14:
No longer present in my version

closed 2018-09-14 by janiversen