subsurface/subsurface #1401: Segfault when opening preferences

p-neves, 2018-06-17:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When opening preferences in Subsurface, I get a segfault

### Operating system:

Arch Linux

### Subsurface version:

Latest master

Yes. Even with the appimage I get the same

4.7.8-356-gc64f421006a4

### Steps to reproduce:

1) Start Subsurface
2) Select File/Preferences

### Current behavior:

Subsurface crashes immediately

### Expected behavior:

Prefences menu opens

### Additional information:

gdb backtrace:

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7fffe2aa7700 (LWP 13881)]
[New Thread 0x7fffe1fff700 (LWP 13882)]
[New Thread 0x7fffe15c5700 (LWP 13883)]
[New Thread 0x7fffd94d9700 (LWP 13884)]
[New Thread 0x7fffc5c04700 (LWP 13885)]
[New Thread 0x7fffbbf8b700 (LWP 13886)]

Thread 1 "subsurface" received signal SIGSEGV, Segmentation fault.
0x00007fffeefaf687 in __strlen_avx2 () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007fffeefaf687 in __strlen_avx2 () at /usr/lib/libc.so.6
#1  0x000055555590d877 in QString::QString(char const*) (ch=0xffffffff <error: Cannot access memory at address 0xffffffff>, this=0x7fffffffd340) at /usr/include/qt/QtCore/qstring.h:693
#2  0x000055555590d877 in PreferencesDefaults::refreshSettings() (this=0x555556534020) at /home/paneves/Diveapps/subsurface/desktop-widgets/preferences/preferences_defaults.cpp:56
#3  0x000055555590281d in PreferencesDialog::refreshPages() (this=0x5555562013a0) at /home/paneves/Diveapps/subsurface/desktop-widgets/preferences/preferencesdialog.cpp:117
#4  0x00005555556928c1 in MainWindow::on_actionPreferences_triggered() (this=<optimized out>) at /home/paneves/Diveapps/subsurface/desktop-widgets/mainwindow.cpp:847
#5  0x000055555573df4a in MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) (_o=0x55555600d3a0, _c=QMetaObject::InvokeMetaMethod, _id=12, _a=0x7fffffffd5b0)
    at /home/paneves/Diveapps/subsurface/build/desktop-widgets/subsurface_interface_autogen/EWIEGA46WW/moc_mainwindow.cpp:357
#6  0x0000555555742a43 in MainWindow::qt_metacall(QMetaObject::Call, int, void**) (this=0x55555600d3a0, _c=QMetaObject::InvokeMetaMethod, _id=12, _a=0x7fffffffd5b0)
    at /home/paneves/Diveapps/subsurface/build/desktop-widgets/subsurface_interface_autogen/EWIEGA46WW/moc_mainwindow.cpp:464
#7  0x00007fffefde2add in QMetaObject::activate(QObject*, int, int, void**) () at /usr/lib/libQt5Core.so.5
#8  0x00007ffff1e43183 in QAction::triggered(bool) () at /usr/lib/libQt5Widgets.so.5
#9  0x00007ffff1e4587a in QAction::activate(QAction::ActionEvent) () at /usr/lib/libQt5Widgets.so.5
#10 0x00007ffff1fca43d in  () at /usr/lib/libQt5Widgets.so.5
#11 0x00007ffff1fd1c32 in  () at /usr/lib/libQt5Widgets.so.5
#12 0x00007ffff1fd2c9c in QMenu::mouseReleaseEvent(QMouseEvent*) () at /usr/lib/libQt5Widgets.so.5
#13 0x00007ffff1e89058 in QWidget::event(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#14 0x00007ffff1fd53ac in QMenu::event(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#15 0x00007ffff1e49984 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#16 0x00007ffff1e5146a in QApplication::notify(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#17 0x00007fffefdb8bc9 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#18 0x00007ffff1e50781 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) () at /usr/lib/libQt5Widgets.so.5
#19 0x00007ffff1ea4757 in  () at /usr/lib/libQt5Widgets.so.5
#20 0x00007ffff1ea6fd5 in  () at /usr/lib/libQt5Widgets.so.5
#21 0x00007ffff1e49984 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#22 0x00007ffff1e5125b in QApplication::notify(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#23 0x00007fffefdb8bc9 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#24 0x00007ffff0ffb596 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /usr/lib/libQt5Gui.so.5
#25 0x00007ffff0ffd516 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () at /usr/lib/libQt5Gui.so.5
#26 0x00007ffff0fd4a5c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Gui.so.5
#27 0x00007fffe0333b3d in  () at /usr/lib/libQt5XcbQpa.so.5
#28 0x00007fffefdb785c in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#29 0x00007fffefdbfb56 in QCoreApplication::exec() () at /usr/lib/libQt5Core.so.5
#30 0x000055555568f0e2 in main (argc=<optimized out>, argv=<optimized out>) at /home/paneves/Diveapps/subsurface/subsurface-desktop-main.cpp:115
#31 0x00007fffeee7806b in __libc_start_main () at /usr/lib/libc.so.6
#32 0x000055555569070a in _start ()

### Mentions:

neolit123, 2018-06-17:
@p-neves 
this segfaults because one of the `tank_info` pointers for `name` ends up being corrupt and that shouldn't happen. one potential cause could be having **a lot** of tanks.

can you please share your divelog? share it with me on private if you'd prefer - neolit123 at gmail.com

p-neves, 2018-06-18:
Hi Lubomir:

Thanks for your help. I've just shared my log with you on Dropbox.

Cheers:

Pedro

neolit123, 2018-06-18:
@p-neves 
thanks for sharing the log.

i wasn't able to make this crash on Windows, but it did crash on Ubuntu.
i will investigate later. :+1:

p-neves, 2018-06-18:
On 18-06-2018 11:55, Lubomir I. Ivanov wrote:
>
> @p-neves <https://github.com/p-neves>
> thanks for sharing the log.
>
> i wasn't able to make this crash on Windows, but it did crash on Ubuntu.
> i will investigate later. 👍
>

Thank you Lubomir for having a look.

Cheers:

Pedro

PR #1407 (2018-06-20): sanitize looping over equipment arrays

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [x] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

3 commits that sanitize loops that iterate global equipment arrays (yikes).
because reading adjacent memory can crash from time to time.

### Changes made:

- use MAX_TANK_INFO and MAX_WS_INFO (new) everywhere

### Related issues:

Fixes #1401

### Additional information:

na

### Release note:
closed 2018-06-20 by dirkhh