subsurface/subsurface #1404: Changing from lacal to cloud DB without loosing all dive entries.

satoschu, 2018-06-18:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

I can´t transfer my in the local database logged dives to the cloud.

### Operating system:

Android 6
### Subsurface version:

4.7.8.369
### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2018-06-18:
Can you explain a bit more what your current situation is, what you are trying to do, and what happens?
I assume you are on an Android device in no-cloud mode and you want to write those dives to the cloud? That should work just fine (unless we managed to break this recently)

satoschu, 2018-06-18:
Hallo Dirk,

da Du mir schreibst schnell auf deutsch, wenn es Dir recht ist.

Ich hatte auf meinem Galaxy Note 3 eine lokale Datenbank mit Tauchgängen, dann habe ich einen Cloud Zugang erstellt und alle Tauchgänge wurden nicht mehr angezeigt. Als ich die Cloud Anbindung wieder beendet habe wurden die Tauchgänge wieder angezeigt.

Mit freundlichen Grüßen,

Tobias Schuster

Von: Dirk Hohndel
Gesendet: Montag, 18. Juni 2018 10:00
An: Subsurface-divelog/subsurface
Cc: satoschu; Author
Betreff: Re: [Subsurface-divelog/subsurface] Changing from lacal to cloud DBwithout loosing all dive entries. (#1404)

Can you explain a bit more what your current situation is, what you are trying to do, and what happens?
I assume you are on an Android device in no-cloud mode and you want to write those dives to the cloud? That should work just fine (unless we managed to break this recently)

dirkhh, 2018-06-18:
Deutsch ist fuer mich oft schwieriger, gerade wenn es ums Programmieren (oder ums Tauchen) geht.

Ich sehe, dass das nicht mehr geht. Ich versuche das Problem zu beheben. Koennte ein paar Tage dauern.

satoschu, 2018-06-18:
Okay, thank you for your help! In future i will communicate in english language :-)

commit 7ec9c206 (2018-06-20): QML UI: store nocloud data when adding cloud credentials

We want to allow people to keep dives they collected without a cloud
account. The code was mostly there, we just got confused about the
existing status because we ran through this twice (no cloud -> unknown
-> verified). This way we explicitly remember this kind of transition.

Fixes #1404
closed 2018-06-20 by dirkhh