subsurface/subsurface #1408: Subsurface-mobile beta crash

Ebi129, 2018-06-19:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
When a dive is manually added it is not possible to change depth or duration again. The app crashes when the save button is pressed.

### Operating system:
Android 7.0

### Subsurface version:
mobile 2.1.0(4.7.8.372) beta, official release 2.0.1(4.7.4.280) is o.k.

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

janiversen, 2018-06-19:
problem is in
void fake_dc(struct divecomputer *dc)

	memset(fake, 0, 6 * sizeof(struct sample));

Bad pointer

dirkhh, 2018-06-19:
should be fixed by #1411

closed 2018-06-19 by dirkhh