subsurface/subsurface #1410: mobile: Dive flags are missing from the maps

mfbernardes, 2018-06-19:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Dive flags are missing from the maps.

As reported by John Smith. I can reproduce.

### Operating system:

App Version: 2.1.0 (4.7.8.372)

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

neolit123, 2018-06-19:
@mfbernardes iOS, Android or both?

i think i saw a recent refactoring of icons that might have broken this.

janiversen, 2018-06-19:
interesting there are flags with mobile on Mac, continue testing.

janiversen, 2018-06-19:
testFlight iOS:

error is in MapWidget.qml line 60   cannot open qrc:////dive-location-marker-icon

can the //// be a problem, or is it just the log ?

neolit123, 2018-06-19:
i cannot reproduce this with the mobile-on-desktop build.
`qrc:////` is a potental problem - there is an extra `/` in there.

it's not the log; tested that.

so it could be something in the mobile (iOS?) version so that it has this extra `/` in the QRCs.

janiversen, 2018-06-19:
It sure could, but I am still puzzled because I see flags with mobile-on-desktop (master) but not on iOS (also master).

neolit123, 2018-06-19:
tested on Android with this build from couple of days ago:
https://github.com/Subsurface-divelog/subsurface/releases/tag/continuous

cannot reproduce the bug, so it seems either iOS specific or that particular test flight build has a problem.

janiversen, 2018-06-19:
Your comments gave me an idea, and with #1412 it works on my iPad.

PR #1412 (2018-06-22): mobile: add MapWidget.qrc and solve #1410
move aliases to mobile-resources.qrc

### Describe the pull request:

- [ x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

add MapWidget.qrc to clean mobile-resources.qrc and subsurface.qrc
and adapt builds.

iOS uses only mobile-resources.arc and MapWidget.qrc, whereas
Android uses mobile-resources.qrc, subsurface.qrc and MapWidget.qrc

Desktop uses subsurface.qrc and MapWidget.qrc
mobile uses mobile-resources.qrc [... cut]
closed 2018-06-22 by dirkhh