subsurface/subsurface #1420: Run shellcheck as a part of our CI

glance-, 2018-06-21:
[Shellcheck](https://github.com/koalaman/shellcheck) is a pretty good linter for shellscripts, and we now use quite a few as our build and test infrastructure.

It would be nice if your CI could run shellcheck on those scripts in a take on trying to validate and enforce sane shell scripts.

mikeller, 2026-10-25:
Hi, [PR #5069](https://github.com/subsurface/subsurface/pull/5069) adds ShellCheck to CI for an initial set of shell scripts. Its [ShellCheck check passed](https://github.com/subsurface/subsurface/actions/runs/40147823692), but this change has no application artifact to test: it changes build checks, not Subsurface's behaviour.

If you have time to review the CI approach, could you check whether the initial script list and the workflow's path filters cover the build scripts you had in mind? With ShellCheck installed, you can also run `scripts/run-shellcheck.sh --check` from a checkout of the PR. `scripts/run-shellcheck.sh --report` shows diagnostics in other `.sh` files; findings there are expected and are not yet part of the CI baseline. Please tell us if an important script or change path is missing.

glance-, 2026-10-25:
Fantastic. I took a quick look at the PR and left some suggestions for improvements. Either way it will be a great addition.

closed 2026-10-26 by mikeller