subsurface/subsurface #1422: Subsurface mobile - Caps-loc is standard

satoschu, 2018-06-22:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

If I want to edit a dive or create a new one the caps-loc function on the keyboard is enabled.

### Operating system:

Android 6.0.1

### Subsurface version:

4.7.8.389 and 4.7.8.398

### Steps to reproduce:

Edit a dive or create it

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

janiversen, 2018-06-22:
Problem seem to be Android specific (Qt problem ?), it does not happen on iOS 11.3

dirkhh, 2018-06-22:
I cannot reproduce this on Android 8.1
Currently don't have access to Android 6

satoschu, 2018-06-22:
If I switch off the functionality of the keyboard "Auto capitals - First letter capital in the beginning of every sentence" (in german "Automatische Großschreibung - Den ersten Buchstaben jedes Satzes automatisch groß schreiben.") the failure is gone.

janiversen, 2018-06-23:
That is not subsurface but your android keyboard funcition. The keyboard routine normally includes spelling and correction like making the first letter in a sentence capital. I believe you can adjust this in settings.

satoschu, 2018-06-23:
Bit why this functipn works correctly in all other Apps but not in Subsurface?

dirkhh, 2018-06-23:
Why does it only fail on your phone, not on hundreds of others?
I have that function turned on on my phone - it works perfectly fine here.
So I'm not sure what to tell you...

mikeller, 2026-10-17:
Hi @satoschu,

janiversen and dirkhh couldn't reproduce this on their own Android devices with auto-capitalisation enabled, and you found disabling your Android keyboard's "auto capitals" setting resolved it, that's a general Android keyboard behaviour (capitalising the first letter of a new sentence), not something Subsurface controls.

Closing this as a device/keyboard-configuration quirk rather than a Subsurface bug, since we couldn't reproduce it and no one else has reported it since. Please reopen if you find this happens even with auto-capitalisation disabled.

closed 2026-10-17 by mikeller