subsurface/subsurface #1424: Windows Desktop: Error im Profile creation with long dives

lje, 2018-06-24:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I'm new to Subsurface and want to manually add a couple of dives I did with a simple dive computer that doesn't support data transfer. All of these dives were multi-level, but since I don't have any stored profile I just assume maximum depth at beginning and linear ascent afterwards.
I encountered the following issue: when I try to enter a dive that exceeds a certain length and depth, the generated dive profile does not include any bottom time.

Let's start with what I want to achieve (target profile):
[image]

So I create a new dive, for which Subsurface creates the following default profile:
[image]

In my case, the dive was max depth 15.1 meters and length 55 minutes. If I enter these two values in the left panel, the following (erroneous) dive profile will be generated:
[image]
This profile does now have any bottom time and is very short (about 6 minutes instead of 55 minutes).

Currently I use the following workaroud: I reduce the depth to 12 or 13 meters and enter the length of 55 minutes. This gives the following profile:
[image]

Afterwards I remove the right bottom point to get a linear ascent.
[image]

Now I change back the depth to 15.1 meters in the left panel. This time, Subsurface generates a new profile without the bug:
[image]

Now I remove the right bottom point again (to get a linear ascent) reaching the target profile (first screenshot).

### Operating system:

Windows 7, German

### Subsurface version:

Subsurface 4.7.8, using metric units and German locale

Did not try with other version

Official Windows version

### Steps to reproduce:

1) create new dive
2) enter dive depth of 18 meters and dive time of 59 minutes

### Current behavior:

A dive profile without any bottom time and a very short length (about 6 minutes) will be generated

### Expected behavior:

Generation of a dive of the requested length with appropriate bottom time

### Additional information:

### Mentions:

mturkia, 2018-06-24:
If you want more control over the dive profile, you can add as many waypoints into the graph as you wish. And drag them around on the profile as well. (Double click where you want a waypoint will add one.)

The depth-duration configuration is good if you want the default profile, not when you want to have anything different. Anyway, I fail to reproduce your issue on the development version of Subsurface, didn't try the release. Also, if you prefer inserting the data on numeric format, the dive planner would allow you to do that (well, it only allows full meters / feet).

dirkhh, 2018-06-25:
> I can confirm that the issue is fixed in that version.

Excellent. We are planning to release Subsurface 4.8 soon.

Thanks for confirming that this is fixed!

closed 2018-06-25 by dirkhh