subsurface/subsurface #1427: Switch to QML compiler when building against 5.11 or newer

dirkhh, 2018-06-25:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [x] Discussion request

### Issue long description:

This is basically a reminder to work on this after the 4.8/2.1 release.
As of 5.11 the QML compiler is available which should significantly improve our startup time

### Operating system:

This needs testing on all our platforms for both desktop and mobile

### Subsurface version:

after the 4.8/2.1 release

### Mentions:

@tcanabrava

janmulder, 2018-09-06:
I played around with compiling QML on 5.11. Seems to work. See https://github.com/janmulder/subsurface/tree/qt511-qml-compile for more info. I do not feel this is ready for a pull request.

janiversen, 2018-09-06:
Looking forward to see this in production, I played a bit but died on problems with the C++ interface.

dirkhh, 2018-09-06:
What's working and what isn't?

janmulder, 2018-09-06:
@dirkhh: not sure its a question for me, but see the commit message. For me, its seems to work, but it does not seem to improve speedup time.

dirkhh, 2018-09-06:
Yes, that was a question for you @janmulder - sorry, I was typing on the phone, so I didn't try to at-tag you.
I'm rather disappointed that it doesn't provide speedup. My question was mainly to see if anything is still broken and if you saw any improvements. And I guess the answer is no to both.
I still think this is something that we should consider at some point, but it may need more investigation.

janiversen, 2018-09-06:
Are you the generated qmlc gets included and not the originsl qml files....that would at least explain the lack of difference.

I remember having read somewhere (sorry cannot do a search right now) about needs to change the qrc file, but are not sure what it was.

janmulder, 2018-09-06:
@dirkhh: indeed, nothing seems broken, and also, there seems no speed improvement. Obviously, that is somewhat unexpected, so some analysis.

I compared a recent Travis build with my own (with QML compiler) build. Interestingly, when looking to the generated apk, in the Travis build, there are no .qml or .qmlc files that originate from the Subsurface source tree. I expect that all is packed into binary resource like files during the build process. This is exactly the same for my QML compiler based build.

Interestingly, all the standard Qt QML stuff from QtQuick that we use in the mobile app is in the apk files in .qml and .qmlc format. These are in "assets" and are added by androiddeployqt. This is the case in both build styles, and androiddeployqt does not seem to have options to use the QML compiler (yet?). This might answer the questions of @janiversen above (the result of QML compilation is a .cpp file, not .qmlc).

Further, I can see that in the QML compiler style build, that for each qml file in the Subsurface source tree, a .cpp is generated and compiled (and this does not happen in the normal, like Travis, build).

Currently, I do not know where most time is spent opening the app on the device. I'm testing this with a 1050 dive logbook, so it might well be that QML processing is just a small part of the overall processing in this context.

janiversen, 2018-09-06:
if you run TestQML . in the tests directory you will see it makes a lot of qmlc files with 5.11 but not earlier versions...lhowever it. might be the qml compiler can also generate cpp files (apart from the moc cpp file).

mikeller, 2026-10-17:
Hi @dirkhh,

Revisiting your own investigation here: you found enabling Qt 5.11's QML compiler built and ran fine but gave no measurable startup-time improvement, since most of Qt's own QtQuick files bypass it anyway and startup time was likely dominated by other processing (on a 1050-dive logbook). That Qt5.11-era investigation is now moot given the subsequent move to Qt6, which has its own different QML compilation/caching behaviour.

Closing this as concluded-without-benefit, superseded by the Qt6 migration. Please reopen if Qt6's AOT/caching compiler is worth a fresh look for mobile startup time.

closed 2026-10-17 by mikeller