subsurface/subsurface #1429: Please publish the android application on f-droid

benzea, 2018-06-25:
Some people might be using an android phone with the Google Play support installed. I personally use a Sailfish OS device and am in the situation that I cannot download APKs from Google Play Store.

It would be useful if the APK was available through the F-Droid store or also as a direct download.

CC: @kittykat

atdotde, 2018-06-25:
You can get the APK here on GitHub: https://github.com/Subsurface-divelog/subsurface/releases

closed 2018-06-25 by atdotde

benzea, 2018-06-26:
Oh, nice! Thanks for the information.

schmittlauch, 2018-07-19:
@atdotde Is the app working for you? On my Jolla1 (Android API level 4.1) it's crashing on startup.
Can I find the required Android version somewhere? I have no  Play Services of course, if that's the issue.

dirkhh, 2018-07-19:
Subsurface-mobile on Android requires Android 4.4 or newer

benzea, 2018-07-19:
Hmm, yeah, I just tried it finally, and I have the same problem on my Jolla X.

IzzySoft, 2018-10-11:
@atdotde if the releases have an APK attached (the latest has not), it seems to be a debug build. What's keeping you from having it built by F-Droid (without GMS that is)? If you're OK with *having* it there, I gladly assist you with *bringing* it there.