subsurface/subsurface #1430: "Error parsing the samples" in openSuSE and Suunto D6i

vrobolab, 2018-06-25:
### Describe the issue:
- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Importing dive logs from Suunto D6i got broken in openSuSE somewhere between April and June 2018.

### Operating system:
openSuSE Leap 15 64bit
PDC: Suunto D6i Novo

### Subsurface version:
Subsurface-4.7.8-x86_64.AppImage and Subsurface-4.7.7-x86_64.AppImage
it worked on 4.7.7 in April but does not work now in June in both 4.7.7 and 4.7.8.

### Steps to reproduce:
connect the PDC to PC, select "Import from Dive Computer (Ctrl+D)", choose correct device and its tty, and get the error message "Error parsing the samples" like this: https://i.imgur.com/gf1jJbJ.png
(other dives in the log were imported from a different dive computer)

### Current behavior:
only 2 last logs get imported.

### Expected behavior:
import all logs from the dive computer.

### Additional information:
see libdivecomputer log and dumpfile attached. 

### Mentions:

### Attachments:

[subsurface.log](https://github.com/Subsurface-divelog/subsurface/files/2133815/subsurface.log)
[subsurface.zip](https://github.com/Subsurface-divelog/subsurface/files/2133820/subsurface.zip)

dirkhh, 2018-06-25:
VERY often errors like this are actually caused by a weak battery. Downloading causes a significantly higher strain on the battery in many dive computers and even if they work perfectly fine "normally", downloading then fails. Especially since your email seems to indicate that in April you were successful with the 4.7.7 AppImage, but that the same AppImage fails now, I'm thinking that the failure may not be with Subsurface at all (since that AppImage obviously hasn't changed - what's most likely to have changed, IMHO, is the charge state of your battery).

Can you give this a try?

vrobolab, 2018-06-26:
I have used the computer for the half of recommended battery replacement time - about 1 year and had about 100 dives while Suunto recommends changing the battery every 2 years or 200 dives, and the battery still reports 3.0V without backlight and 2.8V with backlight same as 1 year ago when I bought it.
I will follow your suggestion and send the computer for the service, however before that I would like to save the dive logs from it - AFAIR changing the battery will flush computer's memory. 
Any ideas how to get dive logs from Suunto with low battery?

jefdriesen, 2018-07-06:
This is a bug in the libdivecomputer parser that is already fixed in commit 41f50534aa22bd1a56432c188ab3aecac8afe45f. Thus not a low battery problem.

dirkhh, 2018-07-06:
What puzzled me was that it worked in April and now fails with the same AppImage.
The commit that @jefdriesen mentions is included in Subsurface 4.8.
@vrobolab can you try with that and see if it indeed fixes your issue?

jefdriesen, 2018-07-06:
The underlying problem is that the data format changed after a firmware update. So if the firmware was upgraded, then it's possible that the same subsurface version worked fine before, but not after the firmware upgrade. Another possibility is that the feature that changed in size (the gas change event) was not used in @vrobolab's previous dives, and thus did not trigger the bug.

Anyway, if I try to parse the memory dump with the latest libdivecomputer, then there are no errors at all. But if I switch back to an older fix without the bugfix, then some dives fail to parse.

vrobolab, 2018-07-06:
@dirkhh I will try in 2-3 days.
@jefdriesen The firmware was not changed, however there were gas change events since April, but not before, so I think this is the reason. I will try the latest Subsurface version in a few days.

vrobolab, 2018-07-08:
Subsurface 4.8 downloaded all dives without problems! The issue seems to be the libdivecomputer bug with parsing Suunto dives with gas changes.

closed 2018-07-08 by vrobolab