subsurface/subsurface #1434: Log on Mobile with updated info overwritten by older Windows-version when Cloud-sync??? Help

Nikolaj-V, 2018-06-26:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mfbernardes, 2018-06-26:
@Nikolaj-V Based just on the title is hard to even try to figure out what might have happened. 

Could you update this with:
- What exactly did you do and what happened?
- iOS / Android?
- subsurface-mobile version?
- Subsurface version on windows?

If the data was synced to the cloud at some point it should be possible to recover.

Nikolaj-V, 2018-06-26:
Hello,
I did an upgrade to the windows version, got Subsurface version 4.7.8 and imported the latest dives with a hiccup and restarted after trying twice. 
I did a "sync" from the mobile device, which has Subsurface-mobile version 2.0.1(4.7.4.280).
The mobile device discarded all information regarding "site information and GPS-data, entry and exit air levels, buddies, etc...., that was stored in the device and from the former version/synchronization.
The phone imported the latest dive, included the last and overwrote the second-last 30 dives, which now was blank, regarding to the information fields above. 
The Android is a 8.0.0.
Any suggestions how to get the data back?
/Nikolaj

dirkhh, 2018-06-27:
Were the dives and the edits that you made on the mobile version ever saved to the cloud? If yes, then they are still there and recoverable. If not, it may be a bit trickier - not sure.
Please contact me via email (dirk at subsurface-divelog dot org) from the email address that you used for Cloud storage. And in that email state that you are giving me permission to look at your dive data on the server (I never access users' data without their explicit invitation to do so).
Then I can take a look and hopefully tell you more about what may or may not be recoverable.

emonty, 2018-08-19:
I had a similar scenario earlier today that I was going to start looking in to - but I figure I can leave some info here in case it's helpful to anyone.

Sequence of actions was:
* start subsurface-mobile
* sync new dives from dive computer
* select "Dive Management : Manual sync with cloud"

At this point - it seems that updates to the dives made on the laptop and synced to the cloud had not been synced to the mobile device (there was, it seems, a network issue when it tried to sync on startup) The result was that the new dives were discarded and the state from the laptop became reality. (simple solution in this case - re-download from dive computer)

There were no conflicting edits made - new dives on the mobile, edits to old dives on the laptop. So - it certainly wasn't git pull --ff-only ... but it should have been a successful no-conflict merge. Do you think it would be worthwhile working on a patch to attempt a merge in such cases? Or do we not necessarily want to go down the auto-merge path? If we don't want merging, perhaps a dialog warning "syncing with cloud could result in data loss" or something? I'd be happy to work on patches for either.

dirkhh, 2018-08-23:
Sorry for the slow response - I actually was on a family vacation.
We DO try to merge, but we are being pretty stupid about it. So yes, improvements of that code would be very welcome.