subsurface/subsurface #1441: Deletion of images in profile not working

bstoeger, 2018-06-28:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

While trying to test Bluetooth on Windows I found a usability bug: Deleting images does only work sporadically in the profile plot. I can only guess that this has something to do with case insensitive filenames? It's difficult for me to debug as I have no build environment on Windows.

### Operating system:

### Subsurface version:

Windows 7 Enterprise
Version 6.1 (Build 7601: Service Pack 1)

### Steps to reproduce:

Click on trashcan symbol.
### Current behavior:

Nothing.
### Expected behavior:

Picture gets deleted.
### Additional information:

### Mentions:

dirkhh, 2018-06-28:
Since you are a Collaborator... push a branch to GitHub/Subsurface-divelog/subsurface with the code that you want to test. Since you push to the main repo, Travis will kick in, and among other things create a Windows installer for you that will be available as "release" for your branch :-)

bstoeger, 2018-06-28:
Alright - probably not a Windows thing after all, but related to my recent changes to the thumbnails. Weird that so far I've only seen it on Windows. But I think I know how to reproduce.

commit 1de6962e (2018-07-16): Dive pictures: connect DivePictureModel signals in constructor

The DivePictureModel signals were connected int the setProfileState()
function of ProfileWidget2. If this function was called repeatedly
(e.g. after adding a dive), multiple connections were made, which led
to multiply delivered signals. This in turn led to confusion, when
deleting pictures (i.e. multiple pictures were deleted).

Instead, connect once in ProfileWidget2's constructor. It is unclear
whether these signals should be disconnected temporarily if no pictures
are shown. In any case, the weren't, so this shouldn't introduc [... cut]