subsurface/subsurface #1442: mobile beta - download from Mares Puck Pro: unsupported operation

Ebi129, 2018-06-28:
### Describe the issue:

- [ ] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Since a few beta versions Mares dive computers are listed as supported: "Mares: Puck Pro (SERIAL, BLE), Quad (SERIAL, BLE), Quad Air (SERIAL, BLE), Smart (SERIAL, BLE)".
Therefore I tried to download from Puck Pro with my new USB interface. Unfortunately I had no success because nothing happens when I press the Download button. 
The problem could be that the interface has no FTDI chip but a Silicon Labs CP210x controller.

This is the log (shortened):
---------- subsurface.log ----------
Empty filename passed to function
"0.003: Successfully opened logfile /storage/emulated/0/subsurface.log at Thu Jun 28 11:45:05 2018"
"0.004: Starting Subsurface-mobile:2.1.0(4.7.8.435):Android Nougat (7.0):arm:en-DE"
"0.005: built with libdivecomputer v0.7.0-devel-Subsurface-NG (02560a7e7fe82919d584d3edbf3876f90382052c)"
"0.005: built with Qt Version 5.10.1, runtime from Qt Version 5.10.1"
"0.005: built with libgit2 0.26.0"
localBtDevice isn't valid or not connectable
"Created position source android"
"0.021: Created position source android"
"Set GPS service update interval to 300 s"
"0.021: Set GPS service update interval to 300 s"
"0.021: location service is available"
"1.054: Synchronising data file"
"1.072: Load dives from local cache"
"1.128: Successfully opened dive data"
"1.185: AppState changed to active with save ongoing and no unsaved changes"
"1.187: 23 dives loaded from cache"
"1.189: have cloud credentials, but user asked not to connect to network"
"Set GPS service update interval to 300 s"
"1.190: Set GPS service update interval to 300 s"
Using the following font: Roboto
qqwindow devicePixelRatio 3 3
Supported dive computers:
...
"Mares: Puck Pro (SERIAL, BLE), Quad (SERIAL, BLE), Quad Air (SERIAL, BLE), Smart (SERIAL, BLE)"
...
qqwindow screen has ldpi/pdpi 72 147.452
"24.829: DCDownloadThread started for Mares Puck Pro on FTDI"
Starting download from  ftdi
Starting the thread 0
Finishing download thread: "Unable to open ftdi Mares (Puck Pro)"
"24.848: Unsupported operation"
no new dives downloaded
"24.859: DCDownloadThread finished"

---------- libdivecomputer.log ----------
Subsurface: v4.7.8-435-g478b5de9e706, built with libdivecomputer v0.7.0-devel-Subsurface-NG (02560a7e7fe82919d584d3edbf3876f90382052c)
INFO: Open: name=ftdi
ERROR: No such file or directory (2) [in /data/android/subsurface/libdivecomputer/src/serial_posix.c:295 (dc_serial_open)]
---------- finish ----------

### Operating system:

Android 7.0 on Moto G5

### Subsurface version:

mobile beta 2.1.0(4.7.8.435)

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

BTW, there is a misleading german translation. The button "Copy logs to clipboard" in the About screen is translated to "Tauchgang in die Zwischenablage kopieren". This should be translated to "Protokolle in die Zwischenablage kopieren" or something similar.

### Mentions:

dirkhh, 2018-06-28:
Thanks for the note about the note about the translation error - I'll fix that right away. If you find other errors or oddities, PLEASE let us know.

As for the misleading information about the ability to download - yeah, that's really annoying and I haven't figured out what to do about this. We should be able to support those four dive computers via BLE, but not using the SiLabs cable. And I don't know how to make this distinction in the UI.

Worse, for some dive computers different cables are available, using different chips. So the same dive computer may be supported for USB downloading with one cable, but not with another...

Ebi129, 2018-06-28:
Thank you for your instant reply.
If there is no chance to support the Mares usb-interface in near future at least the list of supported dive computers should be changed. AFAIK all Mares usb-interfaces have the SiLabs chip so the "serial" entry should be deleted in the log.
Greetings
Eberhard

closed 2018-06-28 by Ebi129