subsurface/subsurface #1445: Pictures shown when manually adding dives

bstoeger, 2018-06-28:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [x] Discussion request

### Issue long description:

When adding a dive, the dive photo tab shows the pictures of the previously selected dive(s). While this is technically correct (the dive selection is still valid), it seems weird from an UI perspective.

### Operating system:

```
$ uname -a
Linux laptop 4.15.0-23-generic #25-Ubuntu SMP Wed May 23 18:02:16 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```
I doubt that this is OS-dependent.
### Subsurface version:

Current master.

### Steps to reproduce:
1) Go to photos-tab with dive having pictures.
2) CTRL-+.
### Current behavior:

Previous pictures are shown.
### Expected behavior:

No pictures shown, unless new pictures are added manually (?)
### Additional information:

### Mentions:

bstoeger, 2018-06-28:
N.B.: Same is true for statistics tab.

What about disabling all but the "Notes" and "Equipment" tabs during dive adding?

dirkhh, 2018-09-09:
Are you going to send a patch? :-)
It would be nice to have that for 4.8.2

PR #1673 (2018-09-14): Desktop: improve navigation options on edit
### Describe the pull request:

- [x] Bug fix
- [x] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:
As described in the referenced issue, we where able to navigate to nonlogical static pages (like information, statistics. extra data) when adding a dive. These are output style pages that make no sense on edit or add. Further, disable access to some pages when entering edit mode.

Notice that the small change in file mainwindow.cpp [... cut]
commit 5133a38d (2018-09-14): Desktop: disable UI elements that make no sense during editing

As described in the referenced issue, we where able to navigate to
nonlogical static pages (like information, statistics. extra data) when
adding a dive. These are output style pages that make no sense on
edit or add. Further, disable access to some pages when entering edit mode.

Notice that the small change in file mainwindow.cpp is simply
because this this not work at all, and became superfluous any
way.

Fixes: #1445
closed 2018-09-14 by bstoeger