subsurface/subsurface #1446: Desktop: Cylinder Pressure, Weight not accepted on click on "Apply changes"

bstoeger, 2018-06-28:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

In the "Equipment" tab the currently edited value is not saved when clicking on "Apply changes"

### Operating system:

$ uname -a
Linux laptop 4.15.0-23-generic #25-Ubuntu SMP Wed May 23 18:02:16 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

### Subsurface version:

$ ./subsurface -v
Subsurface v4.7.8-441-g26ce183f7c33,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (02560a7e7fe82919d584d3edbf3876f90382052c)
built with Qt Version 5.9.5, runtime from Qt Version 5.9.5
built with libgit2 0.26.0

### Steps to reproduce:

1) Select dive
2) Go to equipment tab
3) Change start pressure or other value
4) Click on "Apply changes" without pressing enter or changing field.
### Current behavior:

The old value reappears.
### Expected behavior:

The new value should be saved.
### Additional information:

### Mentions:

neolit123, 2018-06-28:
yeah, i saw that the other day.
it needs "enter" first, before apply.

neolit123, 2018-07-08:
today i've tried the silly approach of solving this by sending the Enter key to the cylinder / weight widgets  `Accept.. ` is pressed and before the `Accept` logic is processed, but it doesn't work.

tried some other solutions but everything is too messy.
it might be worth it to not touch this, as it's not that much of a bug.

dirkhh, 2018-07-08:
It's just so unintuitive and annoying :-(

neolit123, 2018-07-09:
> It's just so unintuitive and annoying :-(

i'm moving this in the hands of @tcanabrava .
i had another attempt at it last night and couldn't solve it.

after 3-4 hours of playing around, i was reminded how much fun QWidgets are. :-)
maybe the fix is easy, who knows?

bstoeger, 2020-05-03:
The "apply changes" thing doesn't even exist anymore. Close.

closed 2020-05-03 by bstoeger