subsurface/subsurface #1453: Dark Mode in macOS Mojave

tgondorf, 2018-07-02:
When using Dark Mode in macOS Mojave (Public Beta currently), UI is not usable currently.
(Using Light Mode is fine as in previous macOS versions)

### Describe the issue:
- [ ] Bug
- [ ] Change request
- [X] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:
macOS Mojave (Public Beta) on MacBook / Mac Mini

### Subsurface version:
4.8

### Steps to reproduce:
Download & Install macOS Mojave Public Beta
Enable Dark Mode in System Preferences / General
Open SubSurface

### Current behavior:
UI is not usable (various areas are not rendering well) - see attached image
[image]

### Expected behavior:
UI is usable and supporting Dark Mode

### Additional information:
macOS Mojave is currently in Beta and will be released in Fall 2018

### Mentions:

atdotde, 2018-07-02:
It seems this is a general Qt problem. googling for "qt mac darkmode" shows that the qt people are working on this. Let's hope they resolve this before Mojave becomes public.

neolit123, 2018-07-02:
if this is theme-only related, hopefully they backport the fix in older Qt too.

mikeller, 2026-10-17:
Hi @tgondorf,

At the time, this was a widely-reported, immature Qt dark-mode/macOS-theme issue rather than anything Subsurface-specific, and the Qt team was actively working on it. We haven't seen a single report of unusable dark-mode rendering in all the years of macOS and Qt issues since, across several major Qt and macOS versions, so I'm reasonably confident this was resolved by subsequent upstream Qt improvements, though I can't point to one specific fix commit.

Closing this on that basis. Please reopen with your current macOS and build details if dark-mode rendering is still broken for you.

closed 2026-10-17 by mikeller