subsurface/subsurface #1461: Desktop: Crash when opening User Manual

mfbernardes, 2018-07-04:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

On MacOS when opening User Manual subsurface crashes.

It seems to work fine on Ubuntu 18.04 (appimage).

### Operating system:

MacOS 10.13.5

### Subsurface version:

4.8.0

### Steps to reproduce:

Open Subsurface, click Help->User Manual.

### Current behavior:

User manual page pops up and Subsurface crash.

### Expected behavior:

### Additional information:

Not a that useful without debug info, but it shows a null pointer being dereferenced... somewhere:
```
Process 53941 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x0)
    frame #0: 0x0000000000000000
error: memory read failed for 0x0
Target 0: (Subsurface) stopped.
```

Unfortunately at the moment I can't build with User Manual, so just reporting, someone might know what is going on.

### Mentions:

@bstoeger I see you changed somethings related to User Manual recently so you might have an idea about this problem.

bstoeger, 2018-07-04:
> @bstoeger I see you changed somethings related to User Manual recently so you might have an idea about this problem.

Yes, I changed `UserManual`'s base-class from `QWindow` to `QDialog`. This was necessary so that we can pass a parent-object and still have it shown.

It's hard to see how this would cause a crash, but I can't exclude it either. A backtrace with debug info would be very useful as I don't have access to a Mac.

> Unfortunately at the moment I can't build with User Manual

Perhaps Travis produces debug-builds with user manual?

dirkhh, 2018-07-04:
Building with user manual should be easy. I don't think that requires QtWebKit, does it?
Also, in the Mac Crash Report, even without debugging symbols, you typically get a sufficiently detailed stack trace that's usually enough to figure out what's going on.
The good news (snort) is that I can reproduce the crash here...
The bad news is that there really doesn't seem to be a decent stack trace for Thread #0. :-(
```
Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0:
--> 
    __TEXT                 000000010d734000-000000010db76000 [ 4360K] r-x/rwx SM=COW  /Volumes/VOLUME/Subsurface.app/Contents/MacOS/Subsurface

Global Trace Buffer (reverse chronological seconds):
440938.306657 libextension.dylib        	0x00007fff99ce1533 tearing down extension request for pid 56333
440938.306692 libextension.dylib        	0x00007fff99ce1533 tearing down extension request for pid 56332

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread

[...]
Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x00000000ffffffff  rbx: 0x0000000000000000  rcx: 0x00007fff9424337a  rdx: 0x0000000000000000
  rdi: 0x0000000116bb0350  rsi: 0x00000000ffffffff  rbp: 0x0000000000000000  rsp: 0x0000000000000000
   r8: 0x00000000001b1000   r9: 0x0000000116a00000  r10: 0x00007fff942446f2  r11: 0x0000000000000246
  r12: 0x0000000000000000  r13: 0x0000000000000000  r14: 0x0000000000000000  r15: 0x0000000000000000
  rip: 0x0000000000000000  rfl: 0x0000000000010286  cr2: 0x0000000000000000
  
Logical CPU:     4
Error Code:      0x00000014
Trap Number:     14
```

neolit123, 2018-07-04:
> Perhaps Travis produces debug-builds with user manual?
we use `-DNO_USERMANUAL=OFF` in Travis so there is user manual.

>  don't think that requires QtWebKit, does it?

it does require QtWebKit. i see that we package it in the OSX `.app`.

a couple of things to try:

1. revert the patch from @bstoeger just to make sure that this isn't a weird Qt bug:
https://github.com/Subsurface-divelog/subsurface/commit/9611e92cf0d90e4ce1eaf3c3adbaac0f78f4c884#diff-71dbb5492d97d7156a5c6c7d4733064c

2. @mfbernardes please, test if the printing is broken too. if yes, then it means it's a QtWebKit issue.

bstoeger, 2018-07-04:
At the end of `desktop-widgets/usermanual.cpp`, there is some Mac-specific code. What happens if you remove the function bodies?

bstoeger, 2018-07-04:
> revert the patch from @bstoeger just to make sure that this isn't a weird Qt bug:

Yes, but then also revert
d21d42b69117aae04b68ecc9cc2139e034bde146
otherwise the manual will not show.

mfbernardes, 2018-07-04:
build.sh suggests it requires Webkit.

I tried Travis build, but it links with a openssl 1.0.2m and I could not figure out how to downgrade a brew package :(

mfbernardes, 2018-07-04:
@neolit123 Printing works.

Tomorrow I'll take another look on this.

mfbernardes, 2018-07-05:
I tried to cheat and use the cached Qt (5.9.1) used on Travis... and it doesn't crash.

So... looks like I'll really have to build from source.

bstoeger, 2018-07-05:
I have two branches which might be of interest:
1) https://github.com/bstoeger/subsurface/tree/manual2 removes any Mac specific code.
2) https://github.com/bstoeger/subsurface/tree/manual3 reverses the QWidget->QDialog change. 

If neither makes a difference, I'm at a loss.

mfbernardes, 2018-07-06:
@dirkhh I built QtWebkit from source with Qt 5.10.1, built Subsurface and it works.
I built with your Qt 5.9.1 from Travis, it works.

But the binaries you create crash. I even tried the updated "4.8.0" just in case.

The main differences is I'm on MacOS 10.13.5, with the latest Xcode and SDK.

dirkhh, 2018-07-06:
Hrmpf. I'm about to start the project to switch my build environments to Qt 5.11.1
As a result of this everything will once again get re-built. Maybe that will help resolve "whatever it is" that causes my binaries to crash :-(

mfbernardes, 2018-07-08:
@dirkhh based on the backtrace from #1471, I tried a user-manual.html without any <img> and it loads correctly. As soon as I add any <img> tag with a valid src it crashes.

I'm assuming the underlaying problem of #1417 is the same, as soon as facebook returns an html with images it crashes.

Still, when I build QtWebkit on my macOS 10.13 it works.

I'm not sure if you're using specific arguments to build. I just used:
Tools/Scripts/build-webkit --qt --debug --install --prefix=/Users/bernarde/Qt/5.11.1/clang_64/

dirkhh, 2018-07-08:
I'm turning off video and audio to make it smaller. Which tag / branch do you build?

mfbernardes, 2018-07-08:
branch "qtwebkit-5.212" as you mentioned on the call.

dirkhh, 2018-07-08:
That's the one I am building. So just because - let me build it with all component in case that is what causes the problem :-(

dirkhh, 2018-07-08:
And of course I'm building on macOS 10.11 with Xcode 8.2.1 so that users on macOS 10.11 and newer can use the binaries... build kicked off, I'll test in a couple of hours when it's finished

dirkhh, 2018-07-08:
OK, built 5.212 with the exact same command line (ok, module install path) here on macOS 10.11 with Xcode 8.2.1 and that fails for both Facebook and user manual. CRUD.

dirkhh, 2018-07-09:
With the help of the QtWebKit maintainer I think this was resolved. I'll post new test binaries in a moment and also updated our pre-build libraries we use on Travis (later today). I'll leave this issue open to make sure this is indeed resolved for everyone - would love feedback for the test binaries if possible.

dirkhh, 2018-07-14:
Fixed in 4.8.1

closed 2018-07-14 by dirkhh