subsurface/subsurface #1464: Error when downloading log from OSTC 2N

MWedekind, 2018-07-05:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Trying to import new files from OSTC 2N via USB ends with message: "Fehler Fehler beim Importieren der Tauchgangsdaten"

### Operating system:

macOS 10.13.5 on iMac and MacBoocAir

### Subsurface version:

4.8.0

### Steps to reproduce:

### Current behavior:

Error message

### Expected behavior:

loading files from OSTC

### Additional information:

Subsurface: v4.8.0, built with libdivecomputer v0.7.0-devel-Subsurface-NG (e97a47cca55973199715df0f818b4955e60d3a31)
INFO: Open: name=/dev/tty.usbserial-A4SHCQSJ
ERROR: No such file or directory (2) [in ../../src/serial_posix.c:295 (dc_serial_open)]

### Mentions:

neolit123, 2018-07-05:
@MWedekind
thank you for the report.

`dc_serial_open()` tries to open a system device and it appears that the requested device `/dev/tty.usbserial-A4SHCQSJ` is missing on your system.

maybe someone else can provide more context on how USB/TTY works on OSX?

> Error occured after update of macos from macOS sierra and updating to Subsurface 4.8.0 at the same time

i don't think it's Subsurface's fault in this case.
can you verify if older Subsurface versions work on this OS?

neolit123, 2018-07-05:
/cc @jefdriesen

mfbernardes, 2018-07-05:
@MWedekind just reinstall the FTDI driver. This should create a new, valid /dev/tty and it should work.

I had this problem with my Oceanic last time I upgraded macOS.

jefdriesen, 2018-07-06:
The fact that you can see the USB device, only tell you the USB device is connected. But in order to be able to use it, that's not enough. The FTDI driver needs to be loaded as well. And if there is no `/dev/tty.usbserial-A4SHCQSJ`, that's not the case. So there is a problem with the FTDI driver.

mfbernardes, 2018-07-06:
@MWedekind I see there are a lot of discussions about similar problems with FTDI on MacOS. Some people mention removing the FTDI driver and leaving just the Apple one, others enable drivers kext-dev-mode.

Not sure how to help here. This is driver related, not Subsurface.

MWedekind, 2018-07-06:
@mfbernardes Thx! But, hm, how can I remove the driver? Is there somewhare a description and where can I find the FTDI driver?

mfbernardes, 2018-07-06:
@MWedekind google is your friend. Sorry.

When I updated a simple reinstall worked.

MWedekind, 2018-07-06:
@mfbernardes If I would have found something at google that would bring me foreward I wouldn't have asked... so do you have a link where the deleting of a usb driver is discribed?
I have allready installed the driver several times...without success

But the information in the logfile has changed:
Subsurface: v4.8.0, built with libdivecomputer v0.7.0-devel-Subsurface-NG (e97a47cca55973199715df0f818b4955e60d3a31)
INFO: Open: name=/dev/tty.usbserial-A4SHCQSJ
INFO: Configure: baudrate=115200, databits=8, parity=0, stopbits=0, flowcontrol=0
INFO: Timeout: value=4000
INFO: Sleep: value=100
INFO: Purge: direction=3
INFO: Write: size=1, data=61
INFO: Read: size=6, data=AAAAAAAAAA55
ERROR: Failed to receive the header. [in ../../src/hw_ostc.c:211 (hw_ostc_device_dump)]

mfbernardes, 2018-07-06:
@MWedekind The log now shows the the driver is working! Cool.

Can you please re-download the 4.8.0 and reinstall? Dirk posted a fixed binary, and this should be fixed on there.

dirkhh, 2018-07-07:
Cool. I'm glad this is working for you again!

closed 2018-07-07 by dirkhh