subsurface/subsurface #1468: Integrate Qt 5.11.1 in build environment

janiversen, 2018-07-06:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x ] New feature request
- [ ] Discussion request

### Issue long description:

Test Qt 5.11.1 and especially the ne qml compiler

### Operating system:

All platforms.

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

mfbernardes, 2018-07-06:
Basically the same as #1427

janiversen, 2018-07-06:
Closed due to #1427. I wanted to test the qml compiler special, because I have now heard from several places, that the compiler can introduce UI bugs you have not seen.

dirkhh, 2018-07-07:
Why would you close this? Those two are different. One is about the QML compiler, the other one about updating all my build environments to 5.11.1

neolit123, 2018-07-07:
i'm concerned about doing Qt version bumps while being between a `.0` and a `.1` release.

neolit123, 2018-07-07:
> can you tell me why that last one was 4.8.0 and not 4.7.9 ?

no, i cannot. :)
the problem i see here is that the `.1` should be a fast one that fixes issues present in `.0`. by bumping Qt we could potentially introduce problems that come from Qt. at least that's what i'm seeing with the current state of OSX/iOS bugs.

> I'm happy to call this 4.9...

yep, might as well do a quick MINOR bump.

dirkhh, 2018-07-07:
I'm right now doing fresh builds of latest master against Qt5.10.1 to see which of the errors are indeed new to Qt5.11 :-)

dirkhh, 2018-07-07:
Official 4.8.0 binary based on Qt5.10.1 shows the same Bluetooth error, but then successfully downloads from my Petrel (not Petrel 2 - this is a single stack BT only version... Linus has my Petrel 2 right now).
```Starting download from  BT
Starting the thread 0
IOBluetooth works only on the main thread or a thread with a running CFRunLoop
The connection step took more than expected. Wait another 20 seconds
INFO: dc_device_open error value of 0
```
Going back to the Qt5.11 build after the successful download with Qt5.10, I now get identical behavior: the error message, then a pairing popup from macOS (enter 0000 as PIN) and then a successful download...

janiversen, 2018-09-11:
My build environment works, so closing this.

closed 2018-09-11 by janiversen