subsurface/subsurface #1470: subsufrace and Divecomputer.eu

geny80, 2018-07-07:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

 have Divecomputer (Tecdive). I have subsufrace 4.8 I am trying download dives:
1. I'm clicking File/configure divecomputer
2. Choosing /dev/tty.Bluetooth-Incoming-Port and Connect via bluetooth
3. DiveComputer (D8:80:39:F3:37:7C) Paired
4. when I click "Save", I see on the bottom of subsufrace window "unsupported operation"
5. When Im trying find Divecomputer (tecdive) in "import from divecomputer menu", I can't find this producer - Tecdive"..

### Operating system:

macos High sierra 10.13.5, lang. polish
Divecomputer.eu (tecdive)
from console: 
wartość domyślna	10:40:45.559414 +0200	trustd	cert[2]: AnchorTrusted =(leaf)[force]> 0
wartość domyślna	10:40:45.562254 +0200	com.apple.iCloudHelper	MacOS error: -67050
wartość domyślna	10:40:46.688829 +0200	wirelessproxd	Advert timer expired for client 12 after 2.00 seconds
wartość domyślna	10:40:46.689287 +0200	identityservicesd	Continuity update advertising with data: Advertising request of type 12, priority 4, UseFG 290 (181.25 ms), data <088f698b 85786175 395484e9 1733>, connectable 1, options {
}
wartość domyślna	10:40:46.689518 +0200	wirelessproxd	Updating advert request for client 12 after 2.00 seconds
wartość domyślna	10:40:46.689604 +0200	wirelessproxd	Start advertising for process identityservicesd (332) of type 12 with advertising interval 290 (181.25 ms)
wartość domyślna	10:40:46.689876 +0200	wirelessproxd	advertisingRulesOSX - advertisements: (
        {
        kCBAdvAppleMfgTypeKey = 12;
        kCBAdvDataAppleMfgData = <13ff4c00 0c0e088f 698b8578 61753954 84e91733>;
        kCBAdvOptionUseFGInterval = 0;
    },
        {
        kCBAdvAppleMfgTypeKey = 16;
        kCBAdvDataAppleMfgData = <07ff4c00 10020b00>;
        kCBAdvOptionUseFGInterval = 0;
    }
)
wartość domyślna	10:40:46.690059 +0200	wirelessproxd	Requesting to start advertising for clients 12 16  with (
        {
        kCBAdditionalAppleMfgAdvertisements =         (
                        {
                kCBAdvAppleMfgTypeKey = 16;
                kCBAdvDataAppleMfgData = <07ff4c00 10020b00>;
                kCBAdvOptionUseFGInterval = 0;
            }
        );
        kCBAdvDataAppleMfgData = <13ff4c00 0c0e088f 698b8578 61753954 84e91733>;
        kCBAdvOptionUseFGInterval = 0;
        kCBMsgArgIsPrivilegedDaemon = 1;
    }
)
wartość domyślna	10:40:46.692104 +0200	wirelessproxd	client.trigger:#N Random sample for 0x28000f is skip
wartość domyślna	10:40:46.751124 +0200	wirelessproxd	Started to advertise for type 12 with error (null)
wartość domyślna	10:40:46.751394 +0200	wirelessproxd	Started to advertise for type 16 with error (null)
wartość domyślna	10:40:46.751747 +0200	identityservicesd	calling continuityDidStartAdvertisingOfType:Activity
wartość domyślna	10:40:46.753851 +0200	wirelessproxd	client.trigger:#N Random sample for 0x28000d is skip
wartość domyślna	10:40:46.754464 +0200	wirelessproxd	client.trigger:#N Random sample for 0x28000d is skip
wartość domyślna	10:40:46.773039 +0200	sharingd	Nearby stop advertising of type: 16
wartość domyślna	10:40:46.773183 +0200	sharingd	Nearby start advertising with data: Advertising request of type 16, priority 1, UseFG 432 (270.00 ms), data <0b00>, connectable 1, options {
} priority 0 mode 432 options (null)
wartość domyślna	10:40:46.773486 +0200	wirelessproxd	Stop advertising for process sharingd (351) of type 16
wartość domyślna	10:40:46.773706 +0200	wirelessproxd	advertisingRulesOSX - advertisements: (
        {
        kCBAdvAppleMfgTypeKey = 12;
        kCBAdvDataAppleMfgData = <13ff4c00 0c0e088f 698b8578 61753954 84e91733>;
        kCBAdvOptionUseFGInterval = 0;
    }
[... 329 lines cut ...]
wartość domyślna	10:41:02.448656 +0200	wirelessproxd	client.trigger:#N Random sample for 0x28000f is skip
wartość domyślna	10:41:02.504670 +0200	wirelessproxd	Started to advertise for type 12 with error (null)
wartość domyślna	10:41:02.505187 +0200	wirelessproxd	Started to advertise for type 16 with error (null)
wartość domyślna	10:41:02.505944 +0200	identityservicesd	calling continuityDidStartAdvertisingOfType:Activity
wartość domyślna	10:41:02.511182 +0200	wirelessproxd	client.trigger:#N Random sample for 0x28000d is skip
wartość domyślna	10:41:02.512194 +0200	wirelessproxd	client.trigger:#N Random sample for 0x28000d is skip
wartość domyślna	10:41:02.528333 +0200	wirelessproxd	Started to advertise for type 12 with error (null)
wartość domyślna	10:41:02.528513 +0200	wirelessproxd	Started to advertise for type 16 with error (null)
wartość domyślna	10:41:02.528792 +0200	identityservicesd	calling continuityDidStartAdvertisingOfType:Activity
wartość domyślna	10:41:02.530133 +0200	wirelessproxd	client.trigger:#N Random sample for 0x28000d is skip
wartość domyślna	10:41:02.530705 +0200	wirelessproxd	client.trigger:#N Random sample for 0x28000d is skip

### Subsurface version:

4.8.0.

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

glance-, 2018-07-07:
Subsurface doesn't support configuring anything on the Divecomputer.eu computers. We only support OSTC3-family, OSTC4, OSTC mk2-family and the suunto vyper family. Those are also the only families shown in the menu to the left of the Configure dive computer screen.

I don't own a Mac nor run official binaries, but I can clearly see Tecdiving and DiveComputer.eu in the "Import from Divecomputer" menu.

mfbernardes, 2018-07-07:
@geny80 File->Configure divecomputer has nothing to do with downloading dives.

You should just go to Import->Import from dive computer.
There you have:
[image]

Enabling "Choose Bluetooth download mode" you can Scan, select the dive computer and Save.

You should be able to download then.

If you still have problems, please run it from the cmd line and send us the log.
`/Applications/Subsurface.app/Contents/MacOS/Subsurface -vv`

geny80, 2018-07-07:
Subsurface v4.8.0,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (02560a7e7fe82919d584d3edbf3876f90382052c)
built with Qt Version 5.10.1, runtime from Qt Version 5.10.1
built with libgit2 0.26.0
"validateGL(): created OpenGLContext."
"validateGL(): obtained QOpenGLFunctions."
"validateGL(): detected OpenGL version 2.1."
can't find Qt localization for locale "pl-PL" searching in "/Applications/Subsurface.app/Contents/translations"
Plugins Directory:  QDir( "/Applications" , nameFilters = { "*" },  QDir::SortFlags( Name | IgnoreCase ) , QDir::Filters( Dirs|Files|Drives|AllEntries ) )
loading dive data from ("/Users/macgfpr/Library/Application Support/Subsurface/macgfpr.xml")
Unable to match dive 'program.divelog' (subsurface)
Unable to match dive 'version.divelog' (3)
Unable to match autogroup 'model.divecomputerid.' (TecDiving DiveComputer.eu)
Unable to match autogroup 'deviceid.divecomputerid.' (ffffffff)
Unable to match autogroup 'serial.divecomputerid.d.' (00370037-30345119-38353334)
Set the current dive site: 398220345
Set the current dive site: 398220345

File locations:

cloud URL set as "https://cloud.subsurface-divelog.org//git/....@......[....@...]"
Local git storage: /Users/.../Library/Application Support/Subsurface/cloudstorage/1b87c2ff68927cf9
Cloud URL: https://cloud.subsurface-divelog.org//git/...@....[....@....]
Image hashes: /Users/.../Library/Application Support/Subsurface/hashes
Local picture directory: /Users/..../Library/Application Support/Subsurface/picturedata/

neolit123, 2018-07-07:
^ i don't see an error in the above log.

mfbernardes, 2018-07-08:
@geny80 The log you posted really shows no error. It just shows you could select DiceComputer.eu

If you still having problems to download please run from command line, go through all the steps to download from your dive computer, then the log would have info about what went wrong.

janmulder, 2018-09-20:
#1697 shows that this issue is solved. Close.

closed 2018-09-20 by janmulder