subsurface/subsurface #1471: Mac/Qt5.11: Facebook crash

dirkhh, 2018-07-07:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

### Operating system:

### Subsurface version:

4.8.0.48

### Steps to reproduce:

Try to connect to FB. The login window opens, then Subsurface crashes.

### Current behavior:

Crash after showing
```
Error receiving trust for a CA certificate
Current url call QUrl("https://www.facebook.com/login.php?skip_api_login=1&api_key=427722490709000&signed_next=1&next=https%3A%2F%2Fwww.facebook.com%2Fv2.6%2Fdialog%2Foauth%3Fredirect_uri%3Dhttp%253A%252F%252Fwww.facebook.com%252Fconnect%252Flogin_success.html%26display%3Dpopup%26scope%3Dpublish_actions%252Cuser_photos%26response_type%3Dtoken%252Cgranted_scopes%26client_id%3D427722490709000%26ret%3Dlogin%26logger_id%3Dcac5bb01-a927-1249-7576-fa107c40e2f9&cancel_url=http%3A%2F%2Fwww.facebook.com%2Fconnect%2Flogin_success.html%3Ferror%3Daccess_denied%26error_code%3D200%26error_description%3DPermissions%2Berror%26error_reason%3Duser_denied%23_%3D_&display=popup&locale=en_US&logger_id=cac5bb01-a927-1249-7576-fa107c40e2f9")
Response without access token!
Segmentation fault: 11
```
Crash reporter shows a crash deep in QtWebKit:
```
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_platform.dylib      	0x00007fff86b342ca setjmp + 30
1   org.qt-project.QtWebKit       	0x000000010d1e707c 0x10bc8e000 + 22384764
2   org.qt-project.QtWebKit       	0x000000010d1e6ece 0x10bc8e000 + 22384334
3   org.qt-project.QtWebKit       	0x000000010d2249f2 0x10bc8e000 + 22637042
4   org.qt-project.QtWebKit       	0x000000010d0c2f76 0x10bc8e000 + 21188470
5   org.qt-project.QtWebKit       	0x000000010d0ed82b WebCore::Image::setData(WTF::PassRefPtr<WebCore::SharedBuffer>, bool) + 107
6   org.qt-project.QtWebKit       	0x000000010cfbb3d1 0x10bc8e000 + 20108241
7   org.qt-project.QtWebKit       	0x000000010cfbb5d2 0x10bc8e000 + 20108754
8   org.qt-project.QtWebKit       	0x000000010cfabd1a 0x10bc8e000 + 20045082
9   org.qt-project.QtWebKit       	0x000000010cfabdcc 0x10bc8e000 + 20045260
10  org.qt-project.QtWebKit       	0x000000010cfa7d6c 0x10bc8e000 + 20028780
11  org.qt-project.QtWebKit       	0x000000010d2442ce 0x10bc8e000 + 22766286
12  org.qt-project.QtWebKit       	0x000000010d2428c1 0x10bc8e000 + 22759617
13  org.qt-project.QtWebKit       	0x000000010d244120 0x10bc8e000 + 22765856
14  org.qt-project.QtWebKit       	0x000000010d2431e3 0x10bc8e000 + 22761955
15  org.qt-project.QtCore         	0x000000010f89979a QMetaObject::activate(QObject*, int, int, void**) + 2954
16  org.qt-project.QtNetwork      	0x000000010f5174b2 QNetworkReplyHttpImplPrivate::replyDownloadProgressSlot(long long, long long) + 210
17  org.qt-project.QtCore         	0x000000010f891964 QObject::event(QEvent*) + 788
18  org.qt-project.QtWidgets      	0x000000010e269f12 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 306
19  org.qt-project.QtWidgets      	0x000000010e26b2ed QApplication::notify(QObject*, QEvent*) + 573
20  org.qt-project.QtCore         	0x000000010f86854f QCoreApplication::notifyInternal2(QObject*, QEvent*) + 159
21  org.qt-project.QtCore         	0x000000010f869722 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 850
22  libqcocoa.dylib               	0x0000000112e2a3de QCocoaEventDispatcherPrivate::processPostedEvents() + 190
23  libqcocoa.dylib               	0x0000000112e2ac91 QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void*) + 33
24  com.apple.CoreFoundation      	0x00007fff92f8f7e1 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
25  com.apple.CoreFoundation      	0x00007fff92f6ef0c __CFRunLoopDoSources0 + 556
26  com.apple.CoreFoundation      	0x00007fff92f6e42f __CFRunLoopRun + 927
27  com.apple.CoreFoundation      	0x00007fff92f6de28 CFRunLoopRunSpecific + 296
28  com.apple.HIToolbox           	0x00007fff8bc66935 RunCurrentEventLoopInMode + 235
29  com.apple.HIToolbox           	0x00007fff8bc6676f ReceiveNextEventCommon + 432
30  com.apple.HIToolbox           	0x00007fff8bc665af _BlockUntilNextEventMatchingListInModeWithFilter + 71
31  com.apple.AppKit              	0x00007fff85581df6 _DPSNextEvent + 1067
32  com.apple.AppKit              	0x00007fff85581226 -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
33  libqcocoa.dylib               	0x0000000112e299e7 QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 2231
34  org.qt-project.QtCore         	0x000000010f8640a2 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 418
35  org.qt-project.QtWidgets      	0x000000010e4640e5 QDialog::exec() + 373
36  org.subsurface-divelog        	0x000000010b467757 MainWindow::socialNetworkRequestConnect() + 215
37  org.qt-project.QtCore         	0x000000010f89979a QMetaObject::activate(QObject*, int, int, void**) + 2954
38  org.qt-project.QtWidgets      	0x000000010e2607f5 QAction::activate(QAction::ActionEvent) + 309
39  org.qt-project.QtCore         	0x000000010f891964 QObject::event(QEvent*) + 788
40  org.qt-project.QtWidgets      	0x000000010e2606aa QAction::event(QEvent*) + 234
41  org.qt-project.QtWidgets      	0x000000010e269f12 QApplicationPrivate::notify_helper(QObject*, QEvent*) + 306
42  org.qt-project.QtWidgets      	0x000000010e26b2ed QApplication::notify(QObject*, QEvent*) + 573
43  org.qt-project.QtCore         	0x000000010f86854f QCoreApplication::notifyInternal2(QObject*, QEvent*) + 159
44  org.qt-project.QtCore         	0x000000010f869722 QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 850
45  libqcocoa.dylib               	0x0000000112e2a3de QCocoaEventDispatcherPrivate::processPostedEvents() + 190
46  libqcocoa.dylib               	0x0000000112e2ac91 QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void*) + 33
47  com.apple.CoreFoundation      	0x00007fff92f8f7e1 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
48  com.apple.CoreFoundation      	0x00007fff92f6ef0c __CFRunLoopDoSources0 + 556
49  com.apple.CoreFoundation      	0x00007fff92f6e42f __CFRunLoopRun + 927
50  com.apple.CoreFoundation      	0x00007fff92f6de28 CFRunLoopRunSpecific + 296
51  com.apple.HIToolbox           	0x00007fff8bc66935 RunCurrentEventLoopInMode + 235
52  com.apple.HIToolbox           	0x00007fff8bc6676f ReceiveNextEventCommon + 432
53  com.apple.HIToolbox           	0x00007fff8bc665af _BlockUntilNextEventMatchingListInModeWithFilter + 71
54  com.apple.AppKit              	0x00007fff85581df6 _DPSNextEvent + 1067
55  com.apple.AppKit              	0x00007fff85581226 -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
56  com.apple.AppKit              	0x00007fff85575d80 -[NSApplication run] + 682
57  libqcocoa.dylib               	0x0000000112e29a8d QCocoaEventDispatcher::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 2397
58  org.qt-project.QtCore         	0x000000010f8640a2 QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 418
59  org.qt-project.QtCore         	0x000000010f868c62 QCoreApplication::exec() + 402
60  org.subsurface-divelog        	0x000000010b421b4d main + 5437
```
### Expected behavior:

### Additional information:

### Mentions:

@tcanabrava

dirkhh, 2018-07-07:
Given that the stack trace indicates that all we do is exec() the dialog and the rest happens in QtCore, QtWebKit, and Apple system libraries, I'm not sure this is a Subsurface issue. @thiagomacieira is this something that you have any idea how to poke at?
I built QtWebKit 5.212 from git against Qt5.11.1

thiagomacieira, 2018-07-07:
I wish there was something. It happens deep inside QtWebKit. With debug symbols in that one, we could ask a QtWebKit dev to take a look, to see if it's been fixed.

I did talk to Konstantin this week and he says he'll update QtWebKit to a new version in the coming days. Might be easier to just upgrade.

dirkhh, 2018-07-07:
I can build with debugging easily enough (well - it will take a year). How do I then make sure I link against the debugging version?

neolit123, 2018-07-07:
are we planning for the switch to a newer Qt withing a PATCH version bump of subsurface?
i didn't see which Qt we are using for 4.8.0 on OSX.

given this and the other bluetooth error, i would consider going back to an older version that works.

dirkhh, 2018-07-07:
And just to prove just how few people apparently use the Facebook upload functionality... stock Subsurface 4.8.0 based on Qt5.10.1 shows the same crash with basically identical stack trace. So this is NOT new in Qt5.11...

dirkhh, 2018-07-07:
HRMPF. @thiagomacieira how the heck do I build a debug version of QtWebKit. Apparently running
```
Tools/Scripts/build-webkit --qt --debug --install
```
isn't sufficient. I get a build in `WebKitBuild/Debug` but that doesn't appear to be a debug build (it doesn't have the `_debug` names that I can see with `QtCore` for example:
```
$ ls -l ~/Qt5.11.1/5.11.1/clang_64/lib/QtCore.framework lib
/Users/hohndel/Qt5.11.1/5.11.1/clang_64/lib/QtCore.framework:
total 48
lrwxr-xr-x  1 hohndel  staff    24 Jul  6 09:20 Headers -> Versions/Current/Headers
lrwxr-xr-x  1 hohndel  staff    23 Jul  6 09:20 QtCore -> Versions/Current/QtCore
-rw-r--r--  1 hohndel  staff  1500 Jul  6 09:23 QtCore.prl
lrwxr-xr-x  1 hohndel  staff    29 Jul  6 09:20 QtCore_debug -> Versions/Current/QtCore_debug
-rw-r--r--  1 hohndel  staff  1504 Jul  6 09:23 QtCore_debug.prl
lrwxr-xr-x  1 hohndel  staff    26 Jul  6 09:20 Resources -> Versions/Current/Resources
drwxr-xr-x  4 hohndel  staff   136 Jul  6 09:20 Versions

lib:
total 885648
drwxr-xr-x  6 hohndel  staff        204 Jul  7 09:48 QtWebKit.framework
drwxr-xr-x  6 hohndel  staff        204 Jul  7 09:48 QtWebKitWidgets.framework
-rw-r--r--  1 hohndel  staff    3297320 Jul  7 09:49 libANGLESupport.a
-rw-r--r--  1 hohndel  staff  100919640 Jul  7 09:55 libJavaScriptCore.a
-rw-r--r--  1 hohndel  staff    1927232 Jul  7 09:48 libWTF.a
-rw-r--r--  1 hohndel  staff  281676976 Jul  7 10:13 libWebCore.a
-rw-r--r--  1 hohndel  staff    3599088 Jul  7 10:13 libWebCoreTestSupport.a
-rw-r--r--  1 hohndel  staff   59066368 Jul  7 10:19 libWebKit2.a
-rw-r--r--  1 hohndel  staff    1094928 Jul  7 09:48 libbmalloc.a
-rw-r--r--  1 hohndel  staff     452896 Jul  7 09:48 libbrotli.a
-rwxr-xr-x  1 hohndel  staff     246404 Jul  7 09:49 libgtest.dylib
-rw-r--r--  1 hohndel  staff    1150200 Jul  7 09:48 libwoff2.a
```

thiagomacieira, 2018-07-07:
You need to really want to have a debug build. It takes a lot of RAM and diskspace to build it. There's a magic incantation, but I don't know what it is.

If this isn't urgent, I recommend waiting a few days for a QtWebKit update.

thiagomacieira, 2018-07-07:
Also, you may have got a debug build, just without the `_debug` suffix. It actually makes it easier on the tester, since he wouldn't have to set any environment variables.

dirkhh, 2018-07-07:
I love this. You may have one, you just don't know :-)
Would I expect this build to be SIGNIFICANTLY bigger (as in the other debug builds)?
I'm happy to wait for the update - but I'd also love to be able to be ready to debug if necessary...
What's the best place to talk to the QtWebKit folks these days?

thiagomacieira, 2018-07-07:
Yes, it should be significantly bigger.

Talk to "annulen" on IRC. He's the maintainer, at https://github.com/annulen/webkit

dirkhh, 2018-07-07:
It is indeed significantly bigger :-)
Stack trace:
```
1  setjmp                                                                                                                                                                             0x7fff86b342ca 
2  WebCore::PNGImageReader::decode(WebCore::SharedBuffer const&, bool, unsigned int)                                                                                                  0x1020709a5    
3  WebCore::PNGImageDecoder::decode(bool, unsigned int)                                                                                                                               0x10206f802    
4  WebCore::PNGImageDecoder::isSizeAvailable()                                                                                                                                        0x10206f741    
5  WebCore::ImageSource::isSizeAvailable()                                                                                                                                            0x1020b15b2    
6  WebCore::BitmapImage::isSizeAvailable()                                                                                                                                            0x101f125ed    
7  WebCore::Image::setData(WTF::PassRefPtr<WebCore::SharedBuffer>, bool)                                                                                                              0x101f4db61    
8  WebCore::CachedImage::addIncrementalDataBuffer(WebCore::SharedBuffer&)                                                                                                             0x101dfcb36    
9  WebCore::CachedImage::addDataBuffer(WebCore::SharedBuffer&)                                                                                                                        0x101dfcc51    
10 WebCore::SubresourceLoader::didReceiveDataOrBuffer(const char *, int, WTF::PassRefPtr<WebCore::SharedBuffer>, long long, WebCore::DataPayloadType)                                 0x101de9aca    
11 WebCore::SubresourceLoader::didReceiveBuffer(WTF::PassRefPtr<WebCore::SharedBuffer>, long long, WebCore::DataPayloadType)                                                          0x101de9bf9    
12 WebCore::ResourceLoader::didReceiveBuffer(WebCore::ResourceHandle *, WTF::PassRefPtr<WebCore::SharedBuffer>, int)                                                                  0x101de65cf    
13 WebCore::QNetworkReplyHandler::forwardData()                                                                                                                                       0x1020d2431    
14 WebCore::QNetworkReplyHandlerCallQueue::flush()                                                                                                                                    0x1020d0eef    
15 WebCore::QNetworkReplyWrapper::emitMetaDataChanged()                                                                                                                               0x1020d2149    
16 WebCore::QNetworkReplyWrapper::receiveSniffedMIMEType()                                                                                                                            0x1020d2218    
17 WebCore::QNetworkReplyWrapper::receiveMetaData()                                                                                                                                   0x1020d1728    
18 QMetaObject::activate(QObject *, int, int, void * *)                                                                                                                               0x108d0d79a    
19 QNetworkReplyHttpImplPrivate::replyDownloadProgressSlot(long long, long long)                                                                                                      0x10899d4b2    
20 QObject::event(QEvent *)                                                                                                                                                           0x108d05964    
21 QApplicationPrivate::notify_helper(QObject *, QEvent *)                                                                                                                            0x107744f12    
22 QApplication::notify(QObject *, QEvent *)                                                                                                                                          0x1077462ed    
23 QCoreApplication::notifyInternal2(QObject *, QEvent *)                                                                                                                             0x108cdc54f    
24 QCoreApplicationPrivate::sendPostedEvents(QObject *, int, QThreadData *)                                                                                                           0x108cdd722    
25 QCocoaEventDispatcherPrivate::processPostedEvents()                                                                                                                                0x10fc2a3de    
26 QCocoaEventDispatcherPrivate::postedEventsSourceCallback(void *)                                                                                                                   0x10fc2ac91    
27 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__                                                                                                                         0x7fff92f8f7e1 
[... cut ...]
63 main                                                                                                                                               subsurface-desktop-main.cpp 115 0x10000a09d    
64 start                                                                                                                                                                              0x7fff949a65ad 
65 start                                                                                                                                                                              0x7fff949a65ad 

```
@annulen - any idea what might be causing this?

neolit123, 2018-07-08:
reminder to self to verify this via a more recent AppImage:
https://github.com/Subsurface-divelog/subsurface/issues/783

annulen, 2018-07-09:
This is a known issue: https://github.com/annulen/webkit/wiki/Building-QtWebKit-on-OS-X#troubleshoot

CMake picks up headers from a different libpng installation if it's present, https://github.com/annulen/webkit/wiki/Building-QtWebKit-on-OS-X#warning

dirkhh, 2018-07-09:
Thanks @annulen - I saw this but had no recollection of ever installing Mono. But now that I went hunting for it... yep, it's there. No idea why. Apparently I installed this for "something" more than 3 years ago :-/
I'll remove it and restart the build, hopefully that fixes things.

In somewhat related news, @thiagomacieira mentioned that you are about to do a new QtWebKit release... is there a version that we should/could test in this context?

dirkhh, 2018-07-09:
THANK YOU @annulen - your diagnosis was spot on. Rebuilt after removing Mono and both FB and the user manual HTML view work again.

closed 2018-07-09 by dirkhh