subsurface/subsurface #1472: Mac/Qt5.11: BT download issue

dirkhh, 2018-07-07:
### Describe the issue:

- [ =x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When trying to download from Petrel 2 via BT the user gets a message
```
IOBluetooth works only on the main thread or a thread with a running CFRunLoop
Failed to connect to device  00:13:43:0C:56:29 . Device state  QBluetoothSocket::UnconnectedState . Error:  QBluetoothSocket::SocketError( -2 )
```
### Operating system:

macOS - don't know which version

### Subsurface version:

4.8.0.48

### Steps to reproduce:

Try to download from BT dive computer

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

MarioGeckler, 2018-07-07:
I can´t download dives from a OSTC 2 TR with Subsurface 4.8 on Windows 10 x64 build 1803 also.

When i try to read the divecomputer details i get the message "Unsupported operation"
[image]

I get the error with Subsurface 4.8, with 4.7.8 it works well. Here are the libdivecomputer logs from both versions 
[subsurface_4.7.8.log](https://github.com/Subsurface-divelog/subsurface/files/2172887/subsurface_4.7.8.log)
[subsurface_4.8.log](https://github.com/Subsurface-divelog/subsurface/files/2172888/subsurface_4.8.log)

dirkhh, 2018-07-07:
That is the menu for configuring dive computers, not for downloading from divecomputers.
Please try Importieren->Von Tauchcomputer importieren

neolit123, 2018-07-07:
> IOBluetooth works only on the main thread or a thread with a running CFRunLoop

could be a regression in QtConnectivity. not sure of the timeline here (i.e. in which Qt version this worked for us), but the warning is thrown by code which was added 9 months ago:
https://github.com/qt/qtconnectivity/commit/4bce1481c5cfbd402973c2a488b36499ed44af2e

@ablasche do you know of something that changed in QtConnectivity for OSX recently that could have broke this for us?

i see this `CFRunLoop` check existing since 5.9.3.

@dirkhh what was the previous Qt version where this worked?

dirkhh, 2018-07-07:
Adding the same comment here as in the Qt 5.11 issue (as it really is more relevant here):
Official 4.8.0 binary based on Qt5.10.1 shows the same Bluetooth error, but then successfully downloads from my Petrel (not Petrel 2 - this is a single stack BT only version... Linus has my Petrel 2 right now).
```Starting download from  BT
Starting the thread 0
IOBluetooth works only on the main thread or a thread with a running CFRunLoop
The connection step took more than expected. Wait another 20 seconds
INFO: dc_device_open error value of 0
```
Going back to the Qt5.11 build after the successful download with Qt5.10, I now get identical behavior: the error message, then a pairing popup from macOS (enter 0000 as PIN) and then a successful download...