subsurface/subsurface #1474: Can´t configure a OSTC 2 TR with Subsurface 4.8

MarioGeckler, 2018-07-07:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I can´t configure a OSTC 2 TR with Subsurface 4.8 on Windows.

When i try to connect to the divecomputer over Bluetooth to change settings i get the error message "Unsupported operation"

### Operating system:

Windows 10 x64 build 1803
OSTC 2 TR Firmware 2.97 SP1

### Subsurface version:

The error is in Subsurface 2.8, 2.7.8 is working correctly

### Steps to reproduce:

Try to connect to the divecomputer over Bluetooth to configure it.

### Current behavior:
Error message "Unsupported operation"

### Expected behavior:
Should connect without error message

### Additional information:

[image]

I get the error with Subsurface 4.8, with 4.7.8 it works well. Here are the libdivecomputer logs from both versions 
[subsurface_4.7.8.log](https://github.com/Subsurface-divelog/subsurface/files/2172887/subsurface_4.7.8.log)
[subsurface_4.8.log](https://github.com/Subsurface-divelog/subsurface/files/2172888/subsurface_4.8.log)

### Mentions:

dirkhh, 2018-07-13:
Derek set milestone: 4.8

dirkhh, 2018-07-13:
Derek close: https://github.com/Subsurface-divelog/subsurface/commit/4c2e1529c02bd72ec9662a04f856d9099f3f327d

closed 2018-07-13 by derek[bot]

MarioGeckler, 2018-09-16:
I still get the same error on 4.8.1

Is there any other bluetooth related change from 4.7.8 to 4.8(.1) ?