subsurface/subsurface #1485: TestParse fails on Mac

janiversen, 2018-07-10:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When running TestParse on macOS 10.13 it fails with:

Strange depth reading NaN m
Strange depth reading NaN m
Strange depth reading NaN m
Strange depth reading NaN m
export_dives_xslt with stylesheet uddf-export.xslt
FAIL!  : TestParse::testExport() Compared values are not the same
   Actual   (written.takeFirst().trimmed()): "<depth>2,34</depth>"
   Expected (readin.takeFirst().trimmed()) : "<depth>0</depth>"
   Loc: [/Users/jani/develop/src/subsurface/tests/testparse.cpp(446)]
PASS   : TestParse::parseDL7()
PASS   : TestParse::cleanupTestCase()
Totals: 12 passed, 2 failed, 0 skipped, 0 blacklisted, 3967ms
********* Finished testing of TestParse *********

Seems like a problem.

### Operating system:

### Subsurface version:

### Steps to reproduce:

1) run "make check" in subsurface
2) cd build/tests
3) ./TestParse

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

neolit123, 2018-07-10:
/cc @mturkia

dirkhh, 2018-07-10:
I assume that's with latest master?
Here on macOS 10.11 all tests parse. Oddly, sometimes `TestGpsCoords` crashes when run from `make check`, but when run individually it passes as well. So I wonder what's different about your setup...

mturkia, 2018-07-10:
I'll need at least the testuddfexport.uddf and testuddfexport2.uddf files produced by this test. This might give a hint of what is wrong. However, it does sound weird if this is working on one version of macOS but not on another one. (And I don't have access to any mac so cannot test myself.)

janiversen, 2018-07-10:
my Mac is 10.13.5 and I used a master about 2 days old.  I just rerun "./TraceParse" in build/tests, with a master I updated this morning.

You will find the screen output in TraceParse.log and (I hope all the other generated files).

Remark had to give the file a .txt ending to be accepted, remove that and do a normal tar -xzf.

[tests.tgz.txt](https://github.com/Subsurface-divelog/subsurface/files/2181227/tests.tgz.txt)

mfbernardes, 2018-07-10:
Works correctly here with latest master, on MacOS 10.13.5.
Running manually also pass.

PR #1487 (2018-07-10): UDDF import: add support for decimal comma for depth

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Depth values might apparently contain decimal comma instead of decimal
point with some locale settings. Thus add support for both of them.

### Changes made:

### Related issues:

Fixes #1485

### Additional information:

### Release note:

### Documentation change:
commit 7103f36c (2018-07-10): UDDF import: add support for decimal comma for depth

Depth values might apparently contain decimal comma instead of decimal
point with some locale settings. Thus add support for both of them.

Fixes #1485
closed 2018-07-10 by dirkhh